adafruit/circuitpython #9121: Create host based app to access web workflow file management features

RetiredWizard, 2024-03-30:
it might be nice if there was an application similar to Circup that could be used to manage the file system of Circuit Python devices over a Web Workflow channel.

In addition to streamlining simple operations, this might allow for automated configuration procedures to be used in setting up multiple devices or the use of operating scripts to perform more complicated file management operations.

I'll have to look back at @FoamyGuy's Web Workflow/Circup streams to see if there's a similar approach for creating this application,

FoamyGuy, 2024-05-13:
I'm curious if anyone has ideas on a name for this idea?

I started working on this utility over the weekend, I am in the very beginning stages so far. 

I am envisioning it supporting these commands:

- `ls`: list files and directories on the device optionally filter by the given directory or partial filepath
- `rm`: delete files and directories (with `-r`)  on the device
- `get`: download files and directories from the device to the local computer
- `put`: upload files from the local computer to the device
- `mkdir`: make a new directory with a given name on the device 

I did start this utility by taking circup and stripping out everything not directly related to webworkflow and copying files to the device. Theoretically there exist some opportunity for code re-use if we either made this new utility dependent on circup, or moved the bits that both use into a seperate library and have both the new utility and circup require the new library.  I'm not sure to what extent it's worthwhile though, the circup Backend's contain many more features and functions that this utility won't need, my instinct is to just keep them separate even though there are a few functions  that are technically the same. 

I've tried brainstorming names, but I'm not that happy with anything I've come up with. One thing to decide is whether to make reference to web or web workflow in the name of the utility. I can see the upside of that to make it more apparent that (unlike circup) it does not work with USB Disk workflow. But I could also envision potentially adding support for the BLE workflow which should be technically possible I think, albeit reliant on some BLE library and appropriate hardware in the host. 

Maybe it should reference "wireless" in the name, but not specifically WIFI, web, or web workflow?

I'm also open to any feedback on desired features or functionality for the tool, I do not currently have much in mind beyond the very basics of commands listed above.

dhalbert, 2024-05-13:
wwcli ?

jepler, 2024-05-13:
it would be nice if all workflows could be supported by a single library, which would be used by circup as well as by this new program that lets you ls/cp/rm individual files.

bablokb, 2024-05-14:
I would really like to see `rsync` (based on `put` and `ls`) as an additional command. This command should upload all files changed locally and (like the `--delete` option of `rsync`) remove files on the device that have been deleted locally. You could name it like you like, e.g. `mirror`, but the `rshell`-utility has this `rsync` command as well.

Regarding the name of the utility: I like `wwshell` as proposed by @dhalbert, since this puts the utility in the tradition of tools like `rshell`.

FoamyGuy, 2024-05-20:
I've given this some additional thought and I am thinking about this tentative plan:

Move `backends.py` from the Circup repo into a newly created repo (and probably `shared.py` but need to look closer to confirm). I'm not sure that "Backends" is the best name for it in the context of being it's own library. I am thinking either "circuitpython-connector" or "circuitpython-integration"  both of which are a little more relavent to the libraries purpose in my mind. Backend to me implies that it will be related to some backend server thing like something behind circuitpython.org potentially. The web workflow technically is a backend server but I think "connector" or "integration" terms are better at conveying the idea of the library being a general API for interacting with a connected device.

Once that new library repo exists then submit PR to circup that changes it to use the new library as a dependency and removes backends.py from it's source.

Also once the new library repo exists then a 2nd new repo can be created for the new wwshell (or whatever name we land on) that also depends  on it.

I'm interested in any thoughts or feedback about this plan or the name of the integration library.

RetiredWizard, 2024-05-20:
I really like that plan, thanks for talking this on. I think Circuitpython-connector has a better ring to it :grin:

tannewt, 2024-05-20:
> I'm interested in any thoughts or feedback about this plan or the name of the integration library.

I think it'd be simpler to add a second CLI entry point into this repo instead of splitting it apart. It takes more coordination if it is multiple libraries.

Installing `circup` can just install the new command as well.

RetiredWizard, 2025-10-18:
I believe this has been completed by wwshell.