adafruit/circuitpython #9122: ESP-IDF 5.2.1 not identifying itself as 5.2, but 5.1 instead

bill88t, 2024-03-30:
The following was run with `~/.espressif` wiped.

```
[bill88t@lunchbox | ~/git]> git clone https://github.com/adafruit/circuitpython cptest && cd cptest/ports/espressif && make fetch-port-submodules && cd esp-idf
 [...]
[bill88t@lunchbox | ~/git/cptest/ports/espressif/esp-idf]> git status
HEAD detached at 722fba96a1
nothing to commit, working tree clean
[bill88t@lunchbox | ~/git/cptest/ports/espressif/esp-idf]> ./install.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.11.8
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Updating /home/bill88t/.espressif/idf-env.json
Selected targets are: esp32c2, esp32p4, esp32s2, esp32s3, esp32, esp32c6, esp32h2, esp32c3
Current system platform: linux-arm64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Installing xtensa-esp-elf-gdb@12.1_20231023
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20231023/xtensa-esp-elf-gdb-12.1_20231023-aarch64-linux-gnu.tar.gz
Destination: /home/bill88t/.espressif/dist/xtensa-esp-elf-gdb-12.1_20231023-aarch64-linux-gnu.tar.gz.tmp
Done
Extracting /home/bill88t/.espressif/dist/xtensa-esp-elf-gdb-12.1_20231023-aarch64-linux-gnu.tar.gz to /home/bill88t/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023
Installing riscv32-esp-elf-gdb@12.1_20231023
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20231023/riscv32-esp-elf-gdb-12.1_20231023-aarch64-linux-gnu.tar.gz
Destination: /home/bill88t/.espressif/dist/riscv32-esp-elf-gdb-12.1_20231023-aarch64-linux-gnu.tar.gz.tmp
Done
Extracting /home/bill88t/.espressif/dist/riscv32-esp-elf-gdb-12.1_20231023-aarch64-linux-gnu.tar.gz to /home/bill88t/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023
Installing xtensa-esp-elf@esp-13.2.0_20230928
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-13.2.0_20230928/xtensa-esp-elf-13.2.0_20230928-aarch64-linux-gnu.tar.xz
Destination: /home/bill88t/.espressif/dist/xtensa-esp-elf-13.2.0_20230928-aarch64-linux-gnu.tar.xz.tmp
Done
Extracting /home/bill88t/.espressif/dist/xtensa-esp-elf-13.2.0_20230928-aarch64-linux-gnu.tar.xz to /home/bill88t/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928
Installing riscv32-esp-elf@esp-13.2.0_20230928
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-13.2.0_20230928/riscv32-esp-elf-13.2.0_20230928-aarch64-linux-gnu.tar.xz
Destination: /home/bill88t/.espressif/dist/riscv32-esp-elf-13.2.0_20230928-aarch64-linux-gnu.tar.xz.tmp
Done
Extracting /home/bill88t/.espressif/dist/riscv32-esp-elf-13.2.0_20230928-aarch64-linux-gnu.tar.xz to /home/bill88t/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928
Installing esp32ulp-elf@2.35_20220830
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp32ulp-elf-v2.35_20220830/esp32ulp-elf-2.35_20220830-linux-arm64.tar.gz
Destination: /home/bill88t/.espressif/dist/esp32ulp-elf-2.35_20220830-linux-arm64.tar.gz.tmp
Done
Extracting /home/bill88t/.espressif/dist/esp32ulp-elf-2.35_20220830-linux-arm64.tar.gz to /home/bill88t/.espressif/tools/esp32ulp-elf/2.35_20220830
Installing openocd-esp32@v0.12.0-esp32-20230921
Downloading https://github.com/espressif/openocd-esp32/releases/download/v0.12.0-esp32-20230921/openocd-esp32-linux-arm64-0.12.0-esp32-20230921.tar.gz
Destination: /home/bill88t/.espressif/dist/openocd-esp32-linux-arm64-0.12.0-esp32-20230921.tar.gz.tmp
Done
Extracting /home/bill88t/.espressif/dist/openocd-esp32-linux-arm64-0.12.0-esp32-20230921.tar.gz to /home/bill88t/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921
Installing esp-rom-elfs@20230320
Downloading https://github.com/espressif/esp-rom-elfs/releases/download/20230320/esp-rom-elfs-20230320.tar.gz
Destination: /home/bill88t/.espressif/dist/esp-rom-elfs-20230320.tar.gz.tmp
Done
Extracting /home/bill88t/.espressif/dist/esp-rom-elfs-20230320.tar.gz to /home/bill88t/.espressif/tools/esp-rom-elfs/20230320
Installing Python environment and packages
Creating a new Python environment in /home/bill88t/.espressif/python_env/idf5.1_py3.11_env
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt
Destination: /home/bill88t/.espressif/espidf.constraints.v5.1.txt.tmp
Done
Installing Python packages
 Constraint file: /home/bill88t/.espressif/espidf.constraints.v5.1.txt
 Requirement files:
  - /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in /home/bill88t/.espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 3)) (69.2.0)
Collecting packaging (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 4))
  Using cached https://dl.espressif.com/pypi/packaging/packaging-24.0-py3-none-any.whl (53 kB)
Collecting click (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 7))
  Using cached https://dl.espressif.com/pypi/click/click-8.1.7-py3-none-any.whl (97 kB)
Collecting pyserial (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 8))
  Using cached https://dl.espressif.com/pypi/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting cryptography (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 9))
  Using cached https://dl.espressif.com/pypi/cryptography/cryptography-39.0.2-cp36-abi3-manylinux_2_28_aarch64.whl (4.0 MB)
Collecting pyparsing (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 10))
  Using cached https://dl.espressif.com/pypi/pyparsing/pyparsing-3.0.9-py3-none-any.whl (98 kB)
Collecting pyelftools (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 11))
  Using cached https://dl.espressif.com/pypi/pyelftools/pyelftools-0.29-py2.py3-none-any.whl (174 kB)
Collecting idf-component-manager (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 12))
  Using cached https://dl.espressif.com/pypi/idf-component-manager/idf_component_manager-1.5.2-py2.py3-none-any.whl (160 kB)
Collecting esp-coredump (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 13))
  Using cached https://dl.espressif.com/pypi/esp-coredump/esp_coredump-1.10.0-py3-none-any.whl (44 kB)
Collecting esptool (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 14))
  Using cached https://dl.espressif.com/pypi/esptool/esptool-4.7.0-py3-none-any.whl (370 kB)
Collecting esp-idf-kconfig (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 15))
  Using cached https://dl.espressif.com/pypi/esp-idf-kconfig/esp_idf_kconfig-1.4.2-py3-none-any.whl (44 kB)
Collecting esp-idf-monitor (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 16))
  Using cached https://dl.espressif.com/pypi/esp-idf-monitor/esp_idf_monitor-1.4.0-py3-none-any.whl (44 kB)
Collecting esp-idf-size (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 17))
  Using cached https://dl.espressif.com/pypi/esp-idf-size/esp_idf_size-1.2.0-py3-none-any.whl (45 kB)
Collecting esp-idf-panic-decoder (from -r /home/bill88t/git/cptest/ports/espressif/esp-idf/tools/requirements/requirements.core.txt (line 18))
  Using cached https://dl.espressif.com/pypi/esp-idf-panic-decoder/esp_idf_panic_decoder-1.0.1-py3-none-any.whl (18 kB)
```

Notice `Creating a new Python environment in /home/bill88t/.espressif/python_env/idf5.1_py3.11_env`

This prevents building with an external 5.2.1 idf.

dhalbert, 2024-03-30:
The name for the `python_env` is ESP-IDF's choice, not ours. `main` is building on 5.2.1: you can see comments in `git log` in `esp-idf/` that confirm it is 5.2.1.

> This prevents building with an external 5.2.1 idf.

I don't understand what you mean here.

bill88t, 2024-03-30:
> The name for the python_env is ESP-IDF's choice, not ours.

Cloning espressif's esp-idf release 5.2.1, and running `./install.sh`, shows it correctly identifying itself as 5.2.
The issue is with the forked idf, that is used for CircuitPython.

> I don't understand what you mean here.

Up until now it has been possible to have a seperate esp-idf clone, elsewhere on the system and have it exported to path.
CircuitPython would then build just fine, so long as version matches.
That I refer to as an "external idf", an esp-idf clone outside of the CircuitPython submodules.

However now with the merger of #9071, it's now impossible to do so as CP looks up the stuff it wants through `espidf.constraints.v5.1.txt` not `espidf.constraints.v5.2.txt`.

dhalbert, 2024-03-30:
Aha, I see.

https://github.com/adafruit/esp-idf was missing a bunch of git tags, including `v5.2.1`. I fetch the tags from upstream and pushed them. Locally this seemed to fix the problem. Try again

bill88t, 2024-03-30:
Indeed this fixed the issue, built successfully now. Thanks a lot for the quick fix!

closed 2024-03-30 by bill88t