adafruit/circuitpython #9125: IDF5.2.1 connecting to secure network without a password

bill88t, 2024-03-31:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.0 on 2024-03-28; DFRobot FireBeetle 2 ESP32-S3 with ESP32S3
```

### Code/REPL

```python
import wifi
wifi.radio.connect("secure_network")
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ConnectionError: Unknown failure 210

### Description

Now trying to connect to a secure network with no password produces code 210.

### Additional information

I opened this issue to discuss as to what should we do now.
Do we add a new message or just reuse "Authentication failure"?

Normal authentication failure doesn't produce error 210.

anecdata, 2024-03-31:
210 is new since our definitions:
`NO_AP_FOUND_SECURITY`

bill88t, 2024-03-31:
If you are trying to connect to what you think is an open network and you get "Authentication failure", it's pretty easy to guess what happened. I don't think we need a new message either.

dhalbert, 2024-03-31:
I am making a PR to fix this.

PR #9126 (2024-04-01): handle new ESP-IDF v5.2.1 auth failure codes
- Fixes #9125 

ESP-IDF added some new failure codes for wifi connections. Handle a couple of these by collapsing them to existing "authentication failure" code so they will print with a useful message.

Test: old and new:
```py
Adafruit CircuitPython 9.1.0-beta.0 on 2024-03-28; Adafruit Feather ESP32-S3 Reverse TFT with ESP32S3
>>> import wifi
>>> wifi.radio.connect("MYLOCALNETWORK")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ConnectionError: Unknown failure 210
>>>

Adafruit CircuitPython 9.1.0-beta.0-dirty on 2024-03 [... cut]
closed 2024-04-01 by tannewt