adafruit/circuitpython #9127: raspberrypi Out of MDNS service slots

anecdata, 2024-04-01:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.0 on 2024-03-28; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3

vs.

Adafruit CircuitPython 9.1.0-beta.0 on 2024-03-28; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import time
import os
import traceback
import microcontroller
import wifi
import socketpool
import mdns

PORT1 = 8086
PORT2 = 8088

time.sleep(3)  # wait for serial
print(f'{os.uname()=}')
print(f'{microcontroller.cpu.reset_reason=}')  # is device just coming off a reset?
print(f'{os.getenv("CIRCUITPY_WEB_API_PASSWORD")=}')  # check for web workflow

mdns_server = mdns.Server(wifi.radio)
try:
    print(f"http://{mdns_server.hostname}.local.:{PORT1} starting...")
    mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=PORT1)
    print(f"http://{mdns_server.hostname}.local.:{PORT2} starting...")
    mdns_server.advertise_service(service_type="_fake", protocol="_udp", port=PORT2)
except RuntimeError as ex:
    traceback.print_exception(ex, ex, ex.__traceback__)
```

### Behavior

Fresh reset in each case. Web workflow is not enabled. 

`espressif` allows (as expected) multiple unique service-protocol-port combinations. `raspberrypi` does not.

The only limitation noted in the docs is:
> service_type and protocol can only occur on one port. Any call after the first will update the entry’s port.

https://docs.circuitpython.org/en/latest/shared-bindings/mdns/index.html#mdns.Server.advertise_service

`espressif` output (successful):
```
os.uname()=(sysname='ESP32S3', nodename='ESP32S3', release='9.1.0', version='9.1.0-beta.0 on 2024-03-28', machine='Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3')
microcontroller.cpu.reset_reason=microcontroller.ResetReason.SOFTWARE
os.getenv("CIRCUITPY_WEB_API_PASSWORD")=None
http://cpy-5a9784.local.:8086 starting...
http://cpy-5a9784.local.:8088 starting...
Done.
```

`raspberrypi` output  (unsuccessful):
```
os.uname()=(sysname='rp2040', nodename='rp2040', release='9.1.0', version='9.1.0-beta.0 on 2024-03-28', machine='Raspberry Pi Pico W with rp2040')
microcontroller.cpu.reset_reason=microcontroller.ResetReason.SOFTWARE
os.getenv("CIRCUITPY_WEB_API_PASSWORD")=None
http://cpy-027a20.local.:8086 starting...
http://cpy-027a20.local.:8088 starting...
Traceback (most recent call last):
  File "code.py", line 32, in <module>
RuntimeError: Out of MDNS service slots
Done.
```

### Additional information

(not to be confused with https://github.com/adafruit/circuitpython/issues/8149)

eightycc, 2025-03-22:
`lwip` configuration macro `MDNS_MAX_SERVICES` not specified for `raspberrypi` port, so it was defaulting to 1. Increased to 25, the value specified in the `espressif` port. Results:
```
os.uname()=(sysname='rp2040', nodename='rp2040', release='9.2.5', version='9.2.5-11-ga0b482c280-dirt)
microcontroller.cpu.reset_reason=microcontroller.ResetReason.SOFTWARE
os.getenv("CIRCUITPY_WEB_API_PASSWORD")=jk
http://slots-code-py.local.:8086 starting...
http://slots-code-py.local.:8088 starting...
Done.
```

closed 2025-03-22 by eightycc

anecdata, 2025-03-23:
I just wanted to check if there was any memory impact, and it's <1KB for 25 slots. Haven't (yet) tried running services on all of those slots. Thanks, eightycc!
<details>
<summary>code.py.</summary>

```py
# Adafruit CircuitPython 9.2.6 on 2025-03-23; Pimoroni Pico Plus 2 W with rp2350b
import time
import os
import gc
import random
import traceback
import microcontroller
import wifi
import mdns

BASE_PORT = 8000
NUM_SLOTS = 26  # up to 25 supported, RuntimeError: Out of MDNS service slots on the 26th

def random_service():
    svc = bytearray(b'_')
    for _ in range(0, 8):
        svc.append(random.choice(b'abcdefghijknlmnopqrstuvwxyz'))
    return svc.decode()

time.sleep(3)  # wait for serial
print(f'{os.uname()=}')
print(f'{microcontroller.cpu.reset_reason=}')  # is device just coming off a reset?
print(f'{os.getenv("CIRCUITPY_WEB_API_PASSWORD")=}')  # check for web workflow
gc.collect()
print(f'{gc.mem_free()=}')

mdns_server = mdns.Server(wifi.radio)
try:
    for slot in range(NUM_SLOTS):
        service_type = random_service()
        protocol = '_udp'
        port = BASE_PORT + slot
        print(f"Starting {service_type}.{protocol} on {mdns_server.hostname}.local.:{port}")
        mdns_server.advertise_service(service_type=service_type, protocol=protocol, port=port)
        gc.collect()
        print(f'{gc.mem_free()=}')
except RuntimeError as ex:
    traceback.print_exception(ex, ex, ex.__traceback__)
print("Done.")
```
</details>

PR #10166 (2025-03-22): 9.2.x For raspberrypi, increase MDNS_MAX_SERVICES to 25.
Increases `MDNS_MAX_SERVICES` from default of 1 to 25. This is the value specified for the Espressif port.

Resolves #9127.