adafruit/circuitpython #9129: EPaperDisplay.time_to_refresh is too low to ensure safe refresh

FoamyGuy, 2024-04-01:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0 on 2024-03-19; Adafruit MagTag with ESP32S2
```

### Code/REPL

```python
import time
import keypad
import board

display = board.DISPLAY
buttons = keypad.Keys((board.BUTTON_A, board.BUTTON_B, board.BUTTON_C, board.BUTTON_D), value_when_pressed=False)

while True:
    event = buttons.events.get()
    if event:
        if event.released:
            print(f"btn {event.key_number}")

            try:
                display.refresh()
            except RuntimeError as e:
                print("Caught Runtime error, probably refreshed too soon.")
                print(e)
                print(f"sleeping for: {display.time_to_refresh}")
                time.sleep(display.time_to_refresh)
                display.refresh()
```

### Behavior

If you runthis code and then press two mag tag buttons within quick succession:

The second button press will cause a refresh too soon error which gets caught by the except clause, then we sleep for the value in `display.time_to_refresh` and then try to refresh again. 

Often this ends up just raising another refresh too soon error:
```
code.py output:
btn 0
Caught Runtime error, probably refreshed too soon.
Refresh too soon
sleeping for: 1.367
Traceback (most recent call last):
  File "code.py", line 15, in <module>
RuntimeError: Refresh too soon

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "code.py", line 21, in <module>
RuntimeError: Refresh too soon

Code done running.

```

I found that if I add an extra `0.6` seconds to the sleep then I consistently no longer get the error. Granted this is a bit of a cludge for the workaround, I've only tested on two different devices, it's possible that others have different "magical numbers" required for the additional wait time.

elpekenin, 2024-04-01:
AFAICT the C code works in milliseconds yet it [gets converted](<https://github.com/adafruit/circuitpython/blob/main/shared-bindings/epaperdisplay/EPaperDisplay.c#L300>) to seconds when "passing" it to Py, which probably causes a rounding error which is the real problem making the code fail

I think a viable solution would be to do a **breaking change** so that the property returns in milliseconds and call `sleep_ms` instead of `sleep`

On a side note, maybe there should be some (opt-in) code that automatically flushes the screen (`it time_elapsed_last_draw >= timeout && screen.dirty`) to prevent users from manually messing with the time

tannewt, 2024-04-01:
I think it'd be simpler to add a bit of time to the milliseconds number before it gets divided to float.

PR #9140 (2024-04-03): Fix `EPaperDisplay.time_to_refresh`
Fixes #9129
commit 70c09575 (2024-04-03): fix #9129
closed 2024-04-03 by tannewt