adafruit/circuitpython #9130: usb_hid.set_interface_name() is not working

George-TL, 2024-04-02:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.2 on 2024-03-28; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
# in boot.py file
usb_hid.set_interface_name("HP")
```

### Behavior

USP iInterface field does not change.

### Description

According to pull request https://github.com/adafruit/circuitpython/commit/ffa45dedb1ea306c5419e872b021f6da02f9b7bb usb_hid.set_interface_name() should work under 9.0.2.

### Additional information

@MariuszCwikla what CP version and what hardware was used when testing this feature?

MariuszCwikla, 2024-04-02:
@George-TL are you testing it on windows or on linux?
[linux] can you try running `lsusb -vd 239a:80f4 | grep iInterface` ?
[windows] I have noticed that windows caches name of the HID device. So if you have plugged in your Pico some time ago, then it will store name in registry under `HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_239A&PID_80F4` and basically "ignores" what you set up in `boot.py` (`VID_239A&PID_80F4` are vid/pid of Rasbperry Pico)
Alternatively try changing vid/pid along with `set_interface_name`:

```
supervisor.set_usb_identification(manufacturer='me', product='cool device', vid=0xCAFE, pid=0xBABE)
usb_hid.set_interface_name('cool device')
````

George-TL, 2024-04-02:
Looks like I was drunk and forgot to add "import usb_hid".
Sorry for the noise.

closed 2024-04-02 by George-TL