adafruit/circuitpython #9136: storage.usb_drive_enabled()

p3lim, 2024-04-02:
The storage module provides methods to enable or disable the usb drive, but there is nothing (that I can see) to query if it is already enabled or not. I'm trying to add a toggle for it, so knowing the state would be useful.

dhalbert, 2024-04-02:
You have to do the enabling/disabling in `boot.py`, which runs only after a hard reset, and before USB is set up. You cannot use it in `code.py`. So there is no state to check, since a hard reset resets everything.

You _can_ remember states in, say microcontroller.nvm, or in the file system, if you want to remember the state past a hard-reset or power-cycle.

closed 2024-04-03 by p3lim

dhalbert, 2024-04-03:
The usual technique is to check the state of a button during boot.py. ISo you can hold down the button while pressing reset to avoid the disable code. Here's a discussion: https://learn.adafruit.com/customizing-usb-devices-in-circuitpython/circuitpy-midi-serial#dont-lock-yourself-out-3096636.

dglaude, 2024-04-03:
> The usual technique is to check the state of a button during boot.py. ISo you can hold down the button while pressing reset to avoid the disable code. Here's a discussion: https://learn.adafruit.com/customizing-usb-devices-in-circuitpython/circuitpy-midi-serial#dont-lock-yourself-out-3096636.

Exactly the guide I was trying to locate and the current recommended way to do that.
With the added difficulties that on some hardware, we only have one user button that is also the boot button from the RP2040 like on the QT2040 Trinkey: https://circuitpython.org/board/adafruit_qt2040_trinkey/ so you don't have a button to be used to choose a different behavior (like stealth mode) at boot.

> Edit: was easier than I thought, `microcontroller.nvm` was the solution :)

Could you elaborate on a minimal working example? And in what condition this works (like only for warm reboot?).

I don't know/use `microcontroller.nvm` but I know and use `alarm.sleep_memory[0]` as used in this project: https://learn.adafruit.com/adafruit-magtag-project-selector

So if the guide could be updated with your knowledge/solution, that would be really helpfull.

I want my Trinkey to boot stealth (not CIRCUITPY drive, not REPL, ...) and I want to be able to restart it in "configuration mode" to adapt the code. So if I can use the boot button (after CP has booted) to order a restart in new condition, that would be great.

dhalbert, 2024-04-03:
> My exact usecase is that I use CircuitPython as the firmware for a keyboard, and I have a key combination to trigger that function.

You can also check for a key being held down during reset rather than toggling. You can't use `keypad` to do that, since the key will not transition during boot.py unless you added a delay to monitor it. If it's a key matrix it's a bit tricky: you'll need to make the row be an output or and column be in input, or vice versa, depending on the diodes and how it's wired. But anyway, you can read the key "manually" that way in boot.py.

dhalbert, 2024-04-03:
> I don't know/use `microcontroller.nvm` but I know and use `alarm.sleep_memory[0]` as used in this project: https://learn.adafruit.com/adafruit-magtag-project-selector
> 
> So if the guide could be updated with your knowledge/solution, that would be really helpful.

sleep memory does not survive a power-cycle. `microntroller.nvm` stores bytes in flash. t's pretty easy to use: https://docs.circuitpython.org/en/latest/shared-bindings/nvm/index.html

MrPatben8, 2024-08-16:
I can't get `storage.disable_usb_drive()` to work. I'm running CircuitPython 9.x on a Pi Pico and I've got a boot.py with the following code:
```
import board
import digitalio
import storage

button = digitalio.DigitalInOut(board.GP24)
button.direction = digitalio.Direction.INPUT
button.pull = digitalio.Pull.UP

if button.value:
    storage.disable_usb_drive()
```
    
But regardless of what I try, the USB drive inevitably pops up. Any ideas what could be the problem?