adafruit/circuitpython #9139: 3 KeyboardInterrupt exceptions on boot

TaraDaReal, 2024-04-03:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.2 on 2024-03-28; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import os
import time
import ipaddress
import wifi
import socketpool
import board
import microcontroller
from digitalio import DigitalInOut, Direction
from adafruit_httpserver import Server, Request, Response, POST
import adafruit_requests
import sys

ctrlc = 0

connection = None

ssid = os.getenv('CIRCUITPY_WIFI_SSID')
passwd = os.getenv('CIRCUITPY_WIFI_PASSWORD')

# Configure onboard LED
led = DigitalInOut(board.LED)
led.direction = Direction.OUTPUT
led.value = False

ipv4 =  ipaddress.IPv4Address("192.168.68.198")
netmask =  ipaddress.IPv4Address("255.255.255.0")
gateway =  ipaddress.IPv4Address("192.168.1.1")
wifi.radio.set_ipv4_address(ipv4=ipv4,netmask=netmask,gateway=gateway)

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, "/static",debug=True)

font_family = "monospace"

def is_connected():
    try:
        requests = adafruit_requests.Session(pool)
        response = requests.get("http://www.google.com", timeout=5)
        return True
    except (ValueError, RuntimeError):
        return False
    
def connectToWifi(skipCheck=False):
    if not skipCheck:
        if not is_connected():
            print("Connecting to wifi")
            while not is_connected():
                wifi.radio.connect(ssid, passwd)
                time.sleep(2)
            print(f"Connected to {ssid}")
            return
        return
    wifi.radio.connect(ssid, passwd)

def webpage(state):
    
    html = f"""
            <!DOCTYPE html>
            <html>
            <p>
            
            <form accept-charset="utf-8" method="POST">
            <button class="button" name="TOGGLE LED" value="TOGGLE" type="submit">TOGGLE LED</button>
            </form>
            </p>
            
            <form accept-charset="utf-8" method="POST">
            <button class="button" name="STOP" value="STOP_PICO" type="submit">STOP</button>
            </form>
            </p>
            
            <p>LED is {state}</p>
            </body>
            </html>
            """
    return str(html)

@server.route("/")
def base(request: Request):
    return Response(request, webpage(led.value), content_type='text/html')

@server.route("/", methods=[POST])
def changeLED(request: Request):
    raw_text = request.raw_request.decode('utf-8')
    print(raw_text)
    if "TOGGLE" in raw_text:
        led.value = not led.value
    if "STOP_PICO" in raw_text:
        stopChip()
        
    return Response(request, webpage(led.value), content_type='text/html')

def stopChip():
    print("Stopping")
    time.sleep(3)
    sys.exit()

connectToWifi(True)

print("Starting server")

try:
    server.start(str(wifi.radio.ipv4_address), 80)
    print(f"Listening on {ipv4}:80")
except OSError:
    time.sleep(5)
    print("Restarting")
    microcontroller.reset()
ping_addr = ipaddress.ip_address("8.8.4.4")

clock = time.monotonic()

while True:
    try:
        server.poll()
    except KeyboardInterrupt:
        ctrlc += 1
        if ctrlc == 3:
            print("Stopping because of CTRL-C")
            continue
    except BaseException as e:
        print(e)
        continue
```

### Behavior

```9.0.2\Stopping because of CTRL-C ```
```Code done running.```

The code receives **3** CTRL-C on startup, making the code unable to run without starting it with Thonny or any other IDE.

tannewt, 2024-04-03:
Is thonny started when this happens? Does it work ok with 8.2.x? Do you have anything else connected to the Pico W?

TaraDaReal, 2024-04-07:
Thonny is started, Haven't tried it with 8.2x. It's a barebones Pico W. Only happens on startup, after the issue, when the repl starts CTRL+D or the run button on thonny works fine.

dhalbert, 2024-04-19:
If you don't use Thonny, do you see the same issue?

Stray KeyboardInterrupts sound like Thonny trying to stop a running program.