adafruit/circuitpython #9143: Bluetooth pairing issue on CLUE nRF52840 Express with CircuitPython 9.0.3

TrevKnows, 2024-04-04:
### CircuitPython version

```python
CLUE nRF52840 Express with CircuitPython 9.0.3
```

### Code/REPL

```python
import time
import board
from rainbowio import colorwheel
import neopixel

pixels = neopixel.NeoPixel(board.NEOPIXEL, 10, brightness=0.2, auto_write=False)
rainbow_cycle_demo = 1

def rainbow_cycle(wait):
    for j in range(255):
        for i in range(10):
            rc_index = (i * 256 // 10) + j * 5
            pixels[i] = colorwheel(rc_index & 255)
        pixels.show()
        time.sleep(wait)

while True:
    if rainbow_cycle_demo:
        rainbow_cycle(0.05)
```

### Behavior

The CLUE board does not appear in the list of available Bluetooth devices on the testing device, making it impossible to initiate a pairing process.

The expected behavior of the LED indicating discoverability or pairing mode (solid blue) is not observed; instead, the LED flashes blue, suggesting the board is in a different state or the Bluetooth functionality is not initiating correctly.

### Description

Operating System: macOS Sonoma 14.4.1
Board: Adafruit CLUE nRF52840 Express
CircuitPython Version: 9.0.3
Bluetooth Device for Testing: iPhone, Mac

### Additional information

The file transfer process was completed, but it didn't show that bytes were transferred.
Despite the transfer indication, the boot_out.txt file confirms that the board has been updated to CircuitPython 9.0.3.

tannewt, 2024-04-04:
Did you press reset when the neopixel is blinking blue? Or had you paired to the device already?

tannewt, 2024-04-04:
This isn't a CP bug per se. It is an interaction between the nordic saved word code and the uf2 bootloader version 0.8.2+. 0.8.2 added a splash screen and it uses enough ram that it overwrites our uninitialized section.

PR #9160 (2024-04-09): Move the nRF saved word
Instead of putting it in our uninit section, we need to ensure it is also uninit by the UF2 bootloader. We both leave at least the first 32k of RAM for the soft device, so place our word just before the end of 32k. This is ok because we only use it before the SD is started.

Fixes #9143 and doesn't require a bootloader change.
commit bcae6ae5 (2024-04-09): Move the nRF saved word

Instead of putting it in our uninit section, we need to ensure it
is also uninit by the UF2 bootloader. We both leave at least the
first 32k of RAM for the soft device, so place our word just before
the end of 32k. This is ok because we only use it before the SD is
started.

Fixes #9143 and doesn't require a bootloader change.
PR #9168 (2024-04-11): backport to 9.0.x: Move the nRF saved word
Cherry-picked from https://github.com/adafruit/circuitpython/pull/9160/commits/bcae6ae5cfc989b4c518d98ca2d49961746efb42

From #9160:

> Instead of putting it in our uninit section, we need to ensure it is also uninit by the UF2 bootloader. We both leave at least the first 32k of RAM for the soft device, so place our word just before the end of 32k. This is ok because we only use it before the SD is started.

> Fixes #9143 and doesn't require a bootloader change.
commit 2637e5bf (2024-04-11): Move the nRF saved word

Instead of putting it in our uninit section, we need to ensure it
is also uninit by the UF2 bootloader. We both leave at least the
first 32k of RAM for the soft device, so place our word just before
the end of 32k. This is ok because we only use it before the SD is
started.

Fixes #9143 and doesn't require a bootloader change.
closed 2024-04-09 by dhalbert