adafruit/circuitpython #9154: 9.0.x: nRF: DigitalInOut.value always returns False

kevinjwalters, 2024-04-07:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.3 on 2024-04-04; Adafruit CLUE nRF52840 Express with nRF52840
```

### Code/REPL

```python
Adafruit CircuitPython 9.0.3 on 2024-04-04; Adafruit CLUE nRF52840 Express with nRF52840
>>> import board
>>> import digitalio
>>> import analogio
>>> p0_analogue = analogio.AnalogIn(board.P0)
>>> p0_analogue.value
65523
>>> p0_analogue.deinit()
>>> p0_digital = digitalio.DigitalInOut(board.P0)
>>> p0_digital.value
False
```

### Behavior

As shown, analogue values are ok in 9.0.3 due to https://github.com/adafruit/circuitpython/issues/9109 being fixed by https://github.com/adafruit/circuitpython/pull/9114 but digital values look similarly broken and always return `False`. I have the `3V` pad connected up to `P0` with a wire.

### Description

Probably same issue as analogue issue which sounds like some keen compiler optimisation around memory mapped hardware registers vs use of `volatile`?

dhalbert, 2024-04-07:
There's something really odd going on here. Use of a digital pin gives the expected values _if_ it was not previously used as an analog pin. I'm looking into this.

kevinjwalters, 2024-04-07:
Agreed, I hit reset on CLUE and had a play with P1 (was using P0 before).

```
Adafruit CircuitPython 9.0.3 on 2024-04-04; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>>
>>> import board
>>> import digitalio
>>> p1_digital = digitalio.DigitalInOut(board.P1)
>>> p1_digital.value  ### connected to 3V
True
>>> p1_digital.value  ### over to GND
False
>>> p1_digital.value  ### back to 3V
True
```

dhalbert, 2024-04-08:
It appears that using a pin as an `AnalogIn` messes up any later use as a `DigitalInOut` input. Not only that, but this error is also present in 8.2.10, so it may have been present for quite a while. It is just very rare to switch pin modes like this.

So this is not quite so urgent. Nevertheless, it needs fixing. I did notice in the course of testing that sometimes `AnalogIn` still returns `0` when it should not, but that may be a manifestation of this problem as well.

Test program, tested on Feather nRF52840. Jumper TX to A0.
```py
import board
import digitalio
import analogio
import time

tx = digitalio.DigitalInOut(board.TX)
tx.switch_to_output()

def check_mismatch(x, y):
    return "" if x == y else " -- MISMATCH"

while True:

    a0 = digitalio.DigitalInOut(board.A0)
    a0.switch_to_input()
    print()
    print("DIGITAL: a0 now DigitalInOut")

    tx.value = True
    print(f"{tx.value=}, digital: {a0.value=}{check_mismatch(tx.value, a0.value)}")

    tx.value = False
    print(f"{tx.value=}, digital: {a0.value=}{check_mismatch(tx.value, a0.value)}")

    a0.deinit()
    a0 = analogio.AnalogIn(board.A0)
    print()
    print("ANALOG: a0 now AnalogIn")

    tx.value = True
    print(f"{tx.value=}, analog: {a0.value=}")
    tx.value = False
    print(f"{tx.value=}, analog: {a0.value=}")
    a0.deinit()

    time.sleep(2)
```

dhalbert, 2024-04-10:
Fixed by #9161.

PR #9161 (2024-04-10): nrf: fix regression in AnalogIn, fix problem with reuse as digital pin
Fixes two problems:
- SAADC input channels should be disconnected after SAADC is used, which fixes #9154.
- Regression in 9.0.3: SAADC may return (slightly) negative values when pin is at ground. I took out the logic for that in #9114, not realizing that the SAADC API returns an `int16_t`, not a `uint16_t`.

This means a 9.0.4 release will be done.
closed 2024-04-10 by dhalbert