adafruit/circuitpython #9155: M5Stack CoreS3 with ESP32S3 Camera Not Work

tonyhuang1962, 2024-04-08:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.0 on 2024-03-28; M5Stack CoreS3 with ESP32S3
```

### Code/REPL

```python
import board
import espcamera

cam = espcamera.Camera(
    data_pins=board.CAMERA_DATA,
    external_clock_pin=board.CAMERA_XCLK,
    pixel_clock_pin=board.CAMERA_PCLK,
    vsync_pin=board.CAMERA_VSYNC,
    href_pin=board.CAMERA_HREF,
    pixel_format=espcamera.PixelFormat.JPEG,
    frame_size=espcamera.FrameSize.QQVGA,
    i2c=board.I2C(),
    external_clock_frequency=20_000_000,
 #   grab_mode=espcamera.GrabMode.WHEN_EMPTY
)
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
espidf.IDFError: Requested resource not found

### Description

M5Stack CoreS3 IoT has a 30W pixel camera GC0308  at the bottom of the screen. But seems not work if using the above code. Got   'espidf.IDFError: Requested resource not found' error.

tonyhuang1962, 2024-04-09:
I could not find 9.0.3 on the web. I tried latest 'Adafruit CircuitPython 9.1.0-beta.0-18-gf4dfd5f119 on 2024-04-08; M5Stack CoreS3 with ESP32S3', got the same error 'espidf.IDFError: Requested resource not found'.

dhalbert, 2024-04-09:
This board was added recently, only to `main`, so there is no 9.0.x release for it.

tonyhuang1962, 2024-04-19:
I   added 'CONFIG_GC0308_SUPPORT=y' in sdkconfig to try to enable GC0308 support. But still got 'espidf.IDFError: Requested resource not found' error.

I took the following as reference:
https://github.com/adafruit/circuitpython/issues/8883
https://github.com/adafruit/circuitpython/pull/8901

CDarius, 2024-09-04:
I tested the camera with CircuitPython 9.1.3 and it works. Here a snippet to test it:

```python
import board
import displayio
import struct
import cores3
import adafruit_ticks

cam = cores3.camera(framebuffer_count=2)
display = board.DISPLAY
display_bus = display.bus

display.auto_refresh = False

display_bus.send(42, struct.pack(">hh", 0, 319))
display_bus.send(43, struct.pack(">hh", 0, 239))
t0 = adafruit_ticks.ticks_ms()
while True:
    frame = cam.take(1)
    if isinstance(frame, displayio.Bitmap):
        display_bus.send(44, frame)
        t1 = adafruit_ticks.ticks_ms()
        fps = 1000 / adafruit_ticks.ticks_diff(t1, t0)
        print(f"{fps:3.1f}fps")
        t0 = t1
```

tonyhuang1962, 2024-09-09:
I tested with CircuitPython 9.1.3 but still got the same error.

Adafruit CircuitPython 9.1.3 on 2024-08-30; M5Stack CoreS3 with ESP32S3

>>> 

import board
import espcamera

cam = espcamera.Camera(
    data_pins=board.CAMERA_DATA,
    external_clock_pin=board.CAMERA_XCLK,
    pixel_clock_pin=board.CAMERA_PCLK,
    vsync_pin=board.CAMERA_VSYNC,
    href_pin=board.CAMERA_HREF,
    pixel_format=espcamera.PixelFormat.JPEG,
    frame_size=espcamera.FrameSize.SVGA,
    i2c=board.I2C(),
    external_clock_frequency=20_000_000,
 )

>>>
Traceback (most recent call last):
  File "<stdin>", line 13, in <module>
espidf.IDFError: Operation or feature not supported
>>>

tonyhuang1962, 2024-09-13:
Wow! it works! Thanks CDarius for his great job!

The full example code works like a charm.

closed 2024-09-14 by tonyhuang1962