adafruit/circuitpython #9158: audioio.AudioOut – "All timers in use" on reload

raquo, 2024-04-09:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.2 on Adafruit ItsyBitsy M4 board
```

### Code/REPL

```python
import audioio
import board
import time

time.sleep(1)

print("INIT")

audio = audioio.AudioOut(left_channel=board.A0)

print("DONE!")
```

### Behavior

To trigger the bug, load the provided `code.py` code onto the board, and reload it 4 times (e.g. using Ctrl+D in Mu), letting it the code execute to completion each time. On the **fourth** reload, the board will fail with:

```python
code.py output:
INIT
Traceback (most recent call last):
  File "code.py", line 9, in <module>
RuntimeError: All timers in use
```

It will also fail with the same error on every second reload after this.

Disconnecting and re-connecting the board resets the "counter".

### Description

Seems similar to https://github.com/adafruit/circuitpython/issues/6061 but with AudioOut instead of PWMOut?
Note: A0 pin on ItsyBitsyt M4 has a DAC, so no PWM is involved.

dhalbert, 2024-04-10:
> Note: A0 pin on ItsyBitsyt M4 has a DAC, so no PWM is involved.

We use a TC timer internally to time the DAC conversions. That timer is not being released properly, because there is no finalizer that runs when the `AudioOut` is gc'd. This fix would be part of #8960.

dhalbert, 2024-04-10:
In the meantime you should be able to work around this (if it's a blocker for you), by calling `.deinit()` on the `AudioOut` when the program finishes.

PR #9165 (2024-04-11): switch atmel-samd AudioOut to use finalizer
- Fixes #9158.

Use finalizer for `atmel-samd` `AudioOut` objects. Move `audioio_reset()` logic to `_deinit()`. Also use existing routine to search for a free TC timer.

No other port supports `audioio.AudioOut`. There was an unused mention in the i.mX port.
closed 2024-04-11 by dhalbert