adafruit/circuitpython #9166: busio.UART: read_until() or readline() with definable separator

bludin, 2024-04-10:
Would it be possible to extend busio.UART with a read_until() method as provided by PySerial? Or simply add an optional separator argument to the readline() method?

dhalbert, 2024-04-10:
In principle, it would not be too hard. However, it is pretty easily done in Python: https://github.com/pyserial/pyserial/blob/7aeea35429d15f3eefed10bbb659674638903e3a/serial/serialutil.py#L664. `busio.UART` exists even on the smallest boards, so adding more code uses up firmware space that we might use for functionality that cannot easily be done in Python.

bill88t, 2024-04-14:
Tip: You can use `in_waiting` to read just as many bytes as you want (with `read(no_of_bytes)`) and have a temporary buffer to store stuff.

dhalbert, 2025-02-01:
I'm going to close this given there is a way to do it in Python.

closed 2025-02-01 by dhalbert