adafruit/circuitpython #9177: ESP32 supervisor.reload() does not disconnect user-connected wifi

bill88t, 2024-04-13:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; M5Stack Timer Camera X with ESP32
```

### Code/REPL

```python
Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; M5Stack Timer Camera X with ESP32
>>> import wifi
>>> wifi.radio.connected
False
>>> wifi.radio.connect("Lunchbox", "loli hentai")
>>> wifi.radio.connected
True
>>> # I pressed Ctrl + D here.
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
The `microcontroller` module was used to boot into safe mode.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; M5Stack Timer Camera X with ESP32
>>> import wifi
>>> wifi.radio.connected
False
>>> wifi.radio.connect("Lunchbox", "loli hentai")
>>> wifi.radio.connected
True
>>> import supervisor
>>> supervisor.reload()
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
The `microcontroller` module was used to boot into safe mode.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.0.0-beta.0-52-g6c1e34e957 on 2024-02-17; M5Stack Timer Camera X with ESP32
>>> import wifi
>>> wifi.radio.connected
True
>>>
```

### Behavior

Wifi should have been disconnected, it was not.

### Description

This only happens on plain ESP32 as far as I can tell. I can't repro on s2.

### Additional information

This assumes no wifi auto-connect credentials are installed.

dhalbert, 2024-08-28:
We discussed whether this should happen deliberately or not. The web workflow is available even on user connection. Someone might start using, kind of like user-initiated displays, which stick around past the VM finishing.

bill88t, 2024-08-29:
Well, this isn't linked to web workflow in thos state as it wasn't enabled on that board at all.