adafruit/circuitpython #9179: Block keyboard input during execution of `boot.py` on non-usb boards

bill88t, 2024-04-14:
On usb boards it's not possible to interfere with `boot.py` since usb doesn't start up, however on non-usb boards it's very much possible to do so.

This however makes it possible for serial noise (from plugging the board in) to send a KeyboardInterrupt and stop it.

I think it would eventually be a good idea to outright block all input while `boot.py` is running.

bill88t, 2024-04-14:
This should also perhaps extend to `safemode.py`.

It's pretty much the `boot.py` during safe mode.
Having it be interrupted by a random serial interrupt, leaving the app stopped on a crashed headless board doesn't sound all that fun.

dhalbert, 2024-08-28:
maybe set interrupt character to something non-functional