adafruit/circuitpython #9180: Unexpected Charactes in boot_out.txt

safferyj, 2024-04-14:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.0 on 2024-03-28; Adafruit Feather ESP32-S2 Reverse TFT with ESP32S2
Adafruit CircuitPython 9.1.0-beta.0 on 2024-03-28; FeatherS2 Neo with ESP32S2
```

### Code/REPL

```python
#
```

### Behavior

boot_out.txt file contains 8 unexpected characters betweeen the newline after the UID and the start of "boot.py output". The characters (in hex) are: 1b 5b 32 4b 1b 5b 30 47.

### Description

May be reproduced with no code.py file, and any boot.py file, including an empty boot.py. If there is no boot.py, then there are no unexpected characters. 

### Additional information

[image]

RetiredWizard, 2024-04-14:
Esc [2K is the ansi terminal sequence to erase the current line and Esc [0G places the cursor at column zero.

jepler, 2024-04-14:
```c
    mp_hal_stdout_tx_str(line_clear);
    mp_hal_stdout_tx_str(_current_executing_filename);
    serial_write_compressed(MP_ERROR_TEXT(" output:\n"));
```
The "line clear" code is written by this line in main.c. It was added by @bill88t in 722d8ffca51835b26ae4a032fb5d4795faae974e for https://github.com/adafruit/circuitpython/pull/8725 and this code applies to running both boot.py and code.py. You can check that PR for more background about why it was added; It might be possible to use a different call that bypasses the output capture for boot_out.txt, but I didn't investigate it any further than that.

PR #9181 (2024-04-18): Switch to `serial_write` for the one line clear that is captured by `boot_out.txt`
Fixes #9180.

This is the only function that doesn't get captured into `boot_out.txt`.
I confirmed the line clear still occurs by adding some junk text.

```
Serial console setup
amogusAdafruit CircuitPython 9.1.0-beta.0-26-gc978768d37-dirty on 2024-04-15; M5Stack Timer Camera X with ESP32
Board ID:m5stack_timer_camera_x
UID:806BF178DD44
amogusboot.py output:
Early boot log:
```
The rest of the line clears are outside of what `boot_out.txt` captures and I opted to keep the existing function calls since they seem [... cut]
closed 2024-04-18 by dhalbert