adafruit/circuitpython #9196: STM32F4: Number of endpoints

eidurkr, 2024-04-23:
/ports/stm/mpconfigport.mk

The F4 series receive a blanket number of USB endpoints, but the datasheets reveal a more nuanced story. Shouldn't these be further separated?

0 pairs:
STM32F410xx

4 pairs:
STM32F401xx STM32F405xx STM32F407xx STM32F411xx STM32F415xx STM32F417xx STM32F427xx STM32F437xx STM32F429xx STM32F439xx

6 pairs:
STM32F412xx STM32F413xx STM32F423xx

6+8 pairs:
STM32F446xx STM32F469xx STM32F479xx

dhalbert, 2024-04-23:
These are the F4 chips that are currently in use:
```
STM32F401xD
STM32F401xE
STM32F405RG
STM32F407VG
STM32F411CE
STM32F411xE
STM32F412xGS
STM32F446xx
STM32F746
STM32F767
STM32H743
STM32L4R5ZIY6
```

```sh
ag --nofilename HW_MCU_NAME |awk -e '{print $3;}' | sed s/\"//g | sort -u
```

eidurkr, 2024-04-26:
Good to know, in the meantime at least.

page200, 2025-03-19:
> ```
> STM32F412xGS
> ```

I guess you mean STM32F412xG as per https://www.st.com/resource/en/data_brief/32f412gdiscovery.pdf

Fix proposed in #10150

dhalbert, 2025-03-19:
@page200 I think you're right. The MCU name was added by the person who did the board a long time ago and it just may be a typo. It's not actually used anywhere interesting except that it's included in `os.uname()`, I think.

hierophect, 2025-03-19:
whoops