adafruit/circuitpython #9198: BLE/UART - Nordic dongle stops working - probably entering safe mode

gkecskes78, 2024-04-23:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.4 on 2024-04-16; PCA10059 nRF52840 Dongle with nRF52840
Board ID:pca10059
UID:44AF58CE0AE80785
```

### Code/REPL

```python
import time, board, _bleio, pwmio
from adafruit_ble import BLERadio
from adafruit_ble.services.nordic import UARTService
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement

ble=BLERadio()
uart=UARTService()
advert=ProvideServicesAdvertisement(uart)
_bleio.adapter.enabled=True

time.sleep(2)
print('\n------------------- Start')
print('BLE addr: {}'.format(_bleio.adapter.address))
print('Device name: {}'.format(_bleio.adapter.name))
LED1=pwmio.PWMOut(board.LED1)

connected=False
counter=0
while True:
    if not ble.connected:
        if ble.advertising==False:
            ble.start_advertising(advert)
        if connected==True:
            LED1.duty_cycle=65535
            print("Connection lost")
        connected=False
    if ble.connected:
        if connected==False:
            counter+=1
            LED1.duty_cycle=60000
            print("\nConnection Nr. "+str(counter)+" established")
        for b in ble.connections:
            if connected==False:
                print("ble.connections: "+str(ble.connections))
        connected=True
```

### Behavior

I am periodically setting up a BLE connection between 2 nordic dongles, both having the same CPO version 9.0.4.
Recently a bug relating setting up a BLE connection and activating UART service was fixed - this is working now perfect.
https://github.com/adafruit/Adafruit_CircuitPython_BLE/issues/192

The current issue now is related to the dongle, which gets connected.
The dongle which gets connected, suddenly stops working after a short time. When it stops working, is random, it could happen after 20 loops, but also after 120. This is a resulting screenshot:
[image]

### Description

might be related to [192](https://github.com/adafruit/Adafruit_CircuitPython_BLE/issues/192)

### Additional information

#########################################################
## Following code I am using on the dongle which connects to the other one

import os, gc
import time
import board
import busio
import _bleio
import pwmio
import storage
import analogio
import microcontroller
from adafruit_ble import BLERadio
from digitalio import DigitalInOut, Direction
from adafruit_mcp230xx.mcp23017 import MCP23017
from adafruit_ble.services.nordic import UARTService
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
import adafruit_ads1x15.ads1015 as ADS

devices=[['127','f7e5fe8e19e8']]

ble=BLERadio()
uart=UARTService()
advert=ProvideServicesAdvertisement(uart)
addr=advert.address
_bleio.adapter.enabled=True
adapt_address=str(_bleio.adapter.address).replace(':','')[9:-1]
for item in devices:
    if item[1]==adapt_address:
        _bleio.adapter.name=item[0]

print()
print("devices:"+str(devices))
print('BLE addr: {}'.format(_bleio.adapter.address))

loop=1
connected = False
while loop <= 500:
    if connected == False:
        print("Scanning ...")
    for adv in ble.start_scan(ProvideServicesAdvertisement,timeout=1):
        print("adv: "+str(adv.address))
        add=str(adv.address).replace(":","")[9:-1]
        if add==devices[0][1]:
            print("Address found !")
            if UARTService not in adv.services:
                continue
            cnt=cnt1=None

            print("loop "+str(loop))
            try:
                cnt=ble.connect(adv)
                if cnt.connected:
                    print("Connected, huraa !")
                connected = True
            except Exception as e:
                print("Error #1 " + str(e))

            if connected == True:
                try:
                    cnt1=cnt[UARTService]
                    loop += 1
                    print(" -> OK, UART service on !")
                    time.sleep(2)
                except Exception as e:
                    print("Error #2 " + str(e))
                try:
                    if cnt.connected:
                        print("Disconnecting...")
                        cnt.disconnect()
                        connected = False
                    time.sleep(2)
                except Exception as e:
                    print("Error #3: "+str(e))
                    time.sleep(1)
            print("Free memory: %s"%str(gc.mem_free()))
            print("Allocated memory: %s"%str(gc.mem_alloc()))

        ble.stop_scan()
        time.sleep(0.1)
        break

gkecskes78, 2024-04-23:
The following code I am using on the other dongle to connect and activate UART:
[main.zip](https://github.com/adafruit/circuitpython/files/15073956/main.zip)

dhalbert, 2024-04-24:
If you remove the `PWMOut` use, does it make any difference?
Do you only have nRF52840 dongles to try, or do you also have, say, a Feather nRF52840? The latter uses external flash for CIRCUITPY.

Style note:
`if x:` in this context can be used instead of `if x == True:`. It is idiomatic in Python. Similarly, one would say `if not x:` instead of `if x==False:`.

gkecskes78, 2024-04-25:
Thanks for the style note hint, I updated my code accordingly.
Unfortunately no difference without PWMOut.
We are working with nordic dongles(PCA10059) only but we have some nordic DK (PCA10056) as well. Could it be helpful to check on them ?

dhalbert, 2024-04-25:
> We are working with nordic dongles(PCA10059) only but we have some nordic DK (PCA10056) as well. Could it be helpful to check on them ?

Yes, please try. The CircuitPython build for PCA10059 uses on-board external flash chips for CIRCUITPY. If there is some issue with the chip-internal-flash CIRCUITPY code, then that would be a clue and differentiator.

gkecskes78, 2024-04-26:
I can confirm the same behaviour on PCA10056, CP version 9.0.4 (pca10056_bootloader-0.8.3_s140_6.1.1) and without PWMOut :(
Let me know if I can help further.

dhalbert, 2024-05-29:
I have reproduced this problem. with a simplified version of your test programs on a Feather and ItsyBitsy nRF52840 I see that it's looping forever in `ble_drv_add_event_handler_entry()`. I think this may be due to insufficient locking around a data structure, but need to look further.

PR #9285 (2024-05-31): nordic/bluetooth/ble_drv.c: gc entire event handler list
- Fixes #9198.

Intermittently, the loop in ` ble_drv_add_event_handler_entry()` which walks down the linked list of existing event handlers, would loop infinitely. The list of handlers had a cycle in it. In addition, the entries in the list were damaged. This made me suspect a gc problem.

Some of the list entries were on the heap, and some were not. If some entry were not reachable during a gc, then that block of storage would have been recycled.

The easiest way to fix this was walk down the list when a gc was requested. I tested this [... cut]
closed 2024-05-31 by tannewt