adafruit/circuitpython #9208: ESP32-C3 Wifi only produces error code 2

bill88t, 2024-04-26:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.1-10-gc92bb9b3cc on 2024-04-25; Maker Go ESP32C3 Supermini with ESP32-C3
```

### Code/REPL

```python
import wifi
wifi.radio.connect("Lunchbox", "[REDACTED]")
```

### Behavior

```py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ConnectionError: Unknown failure 2
```

### Description

Regardless of if the password is right or wrong, the result is the same. Tried two different networks.

### Additional information

Wifi error code 2 corresponds to `WIFI_REASON_AUTH_EXPIRE` which makes no sense.
I suspect this is an sdk misconfiguration, like how back when in #8595 C6 didn't work at all.

bablokb, 2024-04-26:
I can only confirm. I have the QT Py ESP32-C3 and never ever managed to get a connection to my router. Also with older versions of CircuitPython. See also https://forums.adafruit.com/viewtopic.php?t=193769 (very last post).

bill88t, 2024-04-30:
Nevermind, this is a board radio issue.
The ESP32-C3 Supertiny antenna is borked. Soldering a custom antenna fixes it.

I tested on my beetle c3 and that worked fine with latest master.
From there I googled about this specific board and found numerous threads that this board has borked wifi, linking sample.
https://community.home-assistant.io/t/esp32-c3-super-mini-wifi-signal-booster/657661

To verify it's really not CircuitPython, I tested idf5.2.1 `examples/wifi/getting-started/station` and indeed, that too failed before the antenna modifications.

@bablokb I think you are experiencing a different issue, or your board is defective.
A way I found that made it work is using your finger as an antenna. That somehow worked..
To do that, press down on the smd antenna with a finger, touching the antenna contacts.

Closing issue.

closed 2024-04-30 by bill88t

SeanTheITGuy, 2024-04-30:
To further complicate the (now closed) issue, I would toss in that I've used the Supermini C3s for many projects now and have never had an issue connecting it to wifi.  I just built the latest to ensure this issue wasn't something newly introduced:  
```
Adafruit CircuitPython 9.1.0-beta.1-11-g94afcaa48a on 2024-04-30; Maker Go ESP32C3 Supermini with ESP32-C3
>>> 
>>> import wifi
>>> wifi.radio.ipv4_address
192.168.73.185
```