adafruit/circuitpython #9209: m5stack atomS3U pin error

gus7av, 2024-04-26:
### CircuitPython version

```python
main
```

### Code/REPL

```python
not tested - just a typo spotted in source code
```

### Behavior

not tested - just a typo spotted in source code

### Description

I spotted an error in the pins.c file for m5stack atom S3u file: https://github.com/adafruit/circuitpython/blob/main/ports/espressif/boards/m5stack_atoms3u/pins.c

BOTH pin GPIO/D1/A1 and GPIO2/D1/A1 are labelled SCL - I imagine one of them should be called SDA

dhalbert, 2024-04-26:
@n3o59hf I believe you submitted the board def?

n3o59hf, 2024-04-26:
Yes, looks like a typo. Based on https://docs.m5stack.com/en/core/AtomS3U GPIO2 should be SDA

PR #9212 (2024-04-27): Fix typo in AtomS3U pins
G2 should be SDA based on https://docs.m5stack.com/en/core/AtomS3U

Fixes #9209
closed 2024-04-27 by dhalbert