adafruit/circuitpython #9211: Walkmp3rson learn guide does not work with CP 9.x  Freezes and displays USB not recognizable on PC Windows

dwetchells, 2024-04-26:
### CircuitPython version

```python
Adafruit CircuitPython version 9.4.0
```

### Code/REPL

```python
===========================================================================
# SPDX-FileCopyrightText: 2022 John Park and Tod Kurt for Adafruit Industries
# SPDX-License-Identifier: MIT
'''Walkmp3rson digital cassette tape player (ok fine it's just SD cards)'''

import time
import os
import board
import busio
import sdcardio
import storage
import audiomixer
import audiobusio
import audiomp3
from adafruit_neokey.neokey1x4 import NeoKey1x4
from adafruit_seesaw import seesaw, rotaryio
import displayio
import terminalio
from adafruit_display_text import label
from adafruit_st7789 import ST7789
from adafruit_progressbar.progressbar import HorizontalProgressBar
from adafruit_progressbar.verticalprogressbar import VerticalProgressBar

displayio.release_displays()

# SPI for TFT display, and SD Card reader on TFT display
spi = board.SPI()
# display setup
tft_cs = board.D6
tft_dc = board.D9
tft_reset = board.D12
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=tft_reset)
display = ST7789(display_bus, width=320, height=240, rotation=90)

# SD Card setup
sd_cs = board.D13
sdcard = sdcardio.SDCard(spi, sd_cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

# I2C NeoKey setup
i2c = busio.I2C(board.SCL, board.SDA)
neokey = NeoKey1x4(i2c, addr=0x30)
amber = 0x300800
red = 0x900000
green = 0x009000

neokey.pixels.fill(amber)
keys = [
    (neokey, 0, green),
    (neokey, 1, red),
    (neokey, 2, green),
    (neokey, 3, green),
]
#  states for key presses
key_states = [False, False, False, False]

# STEMMA QT Rotary encoder setup
rotary_seesaw = seesaw.Seesaw(i2c, addr=0x36)  # default address is 0x36
encoder = rotaryio.IncrementalEncoder(rotary_seesaw)
last_encoder_pos = 0

# file system setup
mp3s = []
for filename in os.listdir('/sd'):
    if filename.lower().endswith('.mp3') and not filename.startswith('.'):
        mp3s.append("/sd/"+filename)

mp3s.sort()  # sort alphanumerically for mixtape  order, e.g., "1_King_of_Rock.mp3"
for mp3 in mp3s:
    print(mp3)
    
track_number = 0
mp3_filename = mp3s[track_number]
mp3_bytes = os.stat(mp3_filename)[6]  # size in bytes is position 6
mp3_file = open(mp3_filename, "rb")
mp3stream = audiomp3.MP3Decoder(mp3_file)
```

### Behavior

Walkmp3rson program crashes.  Does not display anything on TFT display.  If connected to Windows PC, I get a Windows error message stating the USB drive is not recognizable and disconnect the CIRCUITPY drive from my PC.  If I comment the line  >>> mp3stream = audiomp3.MP3Decoder(mp3_file) <<< then the program continue for until the next error which is mp3stream not defined, naturally.

Thank you for all your help and understanding.  ThinMan

### Description

on a RP2040 Feather used in JP Learn guide Walkmp3rson. Used all parts described in guide. Works fine using CP version 8.12. Found issue with module audiomp3 in CP 9.x. Here below is the example I used to cause the crash.  If I comment out the last line the program continues on.  The code is the same as JP guide.

### Additional information

When this line is commented out the program continues 
mp3stream = audiomp3.MP3Decoder(mp3_file)

dhalbert, 2024-06-03:
@dwetchells If you reduce the program to something like the below, does it still crash? I have removed all the display and control peripheral code. Does it crash on pretty much any MP3 file?

# ... appropriate import statements

# SD Card setup
sd_cs = board.D13
sdcard = sdcardio.SDCard(spi, sd_cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

mp3_filename = "something.mp3" # your first filename

mp3_file = open(mp3_filename, "rb")
mp3stream = audiomp3.MP3Decoder(mp3_file)

dwetchells, 2024-06-03:
Yes.  I tried everything I could think of came to the conclusion that the library was not compatible.  If I went back to 8.x it recognized the sd card. Totally repeatable.

Sent from Mars

dwetchells, 2024-06-03:
One other thing.  It tried it using a nRF52480 and it works with cp 9.   I thinks it a rp2040 issue if that helps.

Sent from Mars

dhalbert, 2024-06-04:
I tried the code you have above, only changing some pin assignments to use a TFT FeatherWing with an SD card socket. I am using an Adafruit Feather RP2040. I do not see the hard crash you see after `mp3stream = audiomp3.MP3Decoder(mp3_file)`. I am using a generic Micro Center 16GB SD card.

I am using the two low-bit rateMP3 files from here:
https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/main/MP3_Playback_RP2040/Pico_Multi_File

If you have an MP3 file that causes a crash, could you point to it or drop it here? You may need to .zip it up. Or if you can't do that, could you give me details about the MP3 bit rate, etc.?

dwetchells, 2024-06-04:
Huh.   I will tear it down and restart.   What pin assignment did you choose.  Right now it just crashes the moment it read the SD card.  I am using 9.0.2 so maybe it got fixed in 9.0 5. Fingers crossed.  Thanks for looking into this.  Not sure why the RP2040 is causing the problem for me and the other MC boards are working

Sent from Mars

https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/main/MP3_Playback_RP2040/Pico_Multi_File

dhalbert, 2024-06-04:
I was using a couple of different TFT FeatherWings: https://www.adafruit.com/search?q=tft+featherwing. These don't actually have the correct display (it shows up as a mirror image), but that doesn't matter -- it's not going to affect the operation of the program, because these displays are write-only. But the TFT FeatherWings have an SD socket too, so that provided the SD card for me without having to wire it up. If you have an SD card FeatherWing or similar, you could try that. Else try to make the connections as short as possible.

Try 9.0.5 or 9.1.0-beta.3, but this sounds more like an SD card problem than a version problem. Also try the .mp3's I pointed to above. It may be an MP3 problem of some kind.

dhalbert, 2024-06-04:
I tested again with a 320kb bitrate MP3 and did not see a crash. I'm going to take this issue of this 9.1.0 milestone for now, but happy to keep debugging.

dwetchells, 2024-06-04:
Hi Dan,   So We picked a new RP2040 Feather board and installed 9.1.3 beta an the required 9.0 libraries.
Using all of what John Parks hardware it still fails.   attached is a photo and the code I am using.
If I go back to CP 8 and its libraries.  It works just fine.  If I use other MCUs they all work except the RP2040.  Just so odd  I haven't tried a different SD socket yet since it works with CP 8.0x  just so odd.  I tried serveral SD Cards as well for alot of vendors including what Adafruit sells.  I've event tried creating a SD directory on the SD card.

Thanks again for your time.
Dale Etchells aka Thinman

Once it runs the last line of code below it crashes
# SD Card setup
sd_cs = board.D13
sdcard = sdcardio.SDCard(spi, sd_cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

dwetchells, 2024-06-15:
Hey Dan,  Still testing.  I have JP project working on a nRF52840 with the sound files from JP project.  Still no luck with using an RP2040.  Waiting on some other feathers to try when they come in.  Thats what I get when I use the US Postal system. but it gets here.  UPS has lost to many of my orders.
Thank again.  I'll try changing the bit rate down.  I would not have thought it would stop the directory display of the SD card but what do I know.
Have a super week..

Dale

dwetchells, 2024-07-03:
Hi Dan,   Question.  Which feather board did you use in your testing.  Did you try a RP2040 board.  I ordered Adafruit's latest RP2040 Logger board to see if I get the same non-responsive results.  Thanks again.  I know you guys are all very busy.
Dale Etchells

dhalbert, 2024-07-03:
@dwetchells [I was using](https://github.com/adafruit/circuitpython/issues/9211#issuecomment-2146414506) an Adafruit Feather RP2040. The RP2040 Adalogger should be essentially the same

dhalbert, 2025-07-29:
Hi - Did you ever figure this out?

dwetchells, 2025-07-29:
Hey Dan,  I had not and shelved it.  Let me get back to it and let you know.

Sent from Mars

dwetchells, 2025-07-29:
Hey Dan,  I may have spoken to soon.  New error.  Both with 9.2.8 and 10 Beta 0

The display paints right up to displaying the name of the song and then it crashes with this error message on line 165 which is this command   audio.play(mixer)

I am using all of John Parks hardware in the learn guide.

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
/sd/01_Bartlebeats_Carpeter_.mp3
/sd/02_Bartlebeats_Small_Selective_.mp3
/sd/03_Bartlebeats_Daisy_.mp3
/sd/04_Bartlebeats_Clay Guys_.mp3
Walkmp3rson
Traceback (most recent call last):
 File "code.py", line 165, in <module>
MemoryError: memory allocation failed, allocating 32768 bytes

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

dwetchells, 2025-07-29:
Hi Dan,   So I did a little more digging and tested other OS and found any higher than 9.2.5 & 10 crashes with the memory error message.  9.2.4 works fine.  Thats about as far as I can go in testing.  So something changed in the audiobusio subroutine.
This was fun and have a great day.
Dale Etchells  aka ThinMan

dhalbert, 2025-07-29:
Thanks! That's very helpful.