adafruit/circuitpython #9213: ESP32 `AnalogIn` object has no `value` property

bill88t, 2024-04-29:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.1-14-g126a1a4154 on 2024-04-29; WeMos LOLIN32 Lite with ESP32
```

### Code/REPL

```python
>>> import board
>>> import analogio
>>> a = analogio.AnalogIn(board.IO27)
>>> a.
deinit          reference_voltage
>>> dir(a)
['__class__', '__enter__', '__exit__', 'deinit', 'reference_voltage']
>>>
```

### Behavior

No `value` property.

dhalbert, 2024-04-29:
This is sort of a dupe of #9079. `AnalogIn` does not work if wifi is on: see #9137. There really is a `.value` property, but something about `dir()` is not finding it. If you turn off web workflow, does `.value` magically appear?

anecdata, 2024-04-29:
ADC1 doesn't work with wifi? A number of ADC1 pins have other functions, but GPIO32 and GPIO33 seem available.

bill88t, 2024-04-29:
Hmm, I had missed this, but something's not quite right. I don't have web-workflow enabled, my os is handling the networking, so it's a user-connected network, which should be disconnected on reload.

Quitting from my os and entering repl:
```py
>>> import wifi
>>> wifi.radio.connected
False
>>> import board
>>> import analogio
>>> a = analogio.AnalogIn(board.IO27)
>>> a.
deinit          reference_voltage
>>> wifi.radio.enabled = False
>>> a.
deinit          reference_voltage
>>> a.value
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
espidf.IDFError: Requested resource not found
>>> 
```
However rebooting to safemode:
```py
>>> import board
>>> import analogio
>>> a = analogio.AnalogIn(board.IO27)
>>> a.
value           deinit          reference_voltage
>>> a.value
5679
>>> 
```

This means that in the same board reset, even if we disconnect from wifi, adc is unavailable.
Perhaps some hack could be applied so that when radio is turned off, something is reset so adc can work.

bill88t, 2024-04-29:
> ADC1 doesn't work with wifi? A number of ADC1 pins have other functions, but GPIO32 and GPIO33 seem available.
> 
> _edit:_ maybe 34 and 35 too. 36 and 39 are Hall effect.

Yea ADC1 works..
This should probably be documented as well.

dhalbert, 2024-04-30:
on ESP32, ADC2 is not available when WiFI is in use. I originally though ADC1 was not connected to any pins, but was used only internally. I think it was my misinterpretation of this table from the ESP32 Technical Reference manual:

[image]

dhalbert, 2024-05-24:
The original problem here is that when `.value` raises an exception, it causes `dir(a)` to not show `.value`. That is a `dir(analogio.AnalogIn)` will show `value`. TThe root cause of this is #2179.

There is a documentation problem that says you can't use AnalogIn at all when wifi is in use on the ESP32. That is wrong. You can't use ADC2, but ADC1 works. I will fix the documentation.