adafruit/circuitpython #9214: Badger 2040: Sometimes halts forever when refreshing the screen

mariobodemann, 2024-04-30:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.1-13-g7fb791023c-dirty on 2024-04-28; Pimoroni Badger 2040 with rp2040

(built from https://github.com/gdg-berlin-android/zepython, meaningful difference: https://github.com/gdg-berlin-android/ZePython/blob/10bf2bafb474783c30970ffb7c2567357697b938/shared-module/epaperdisplay/EPaperDisplay.c#L210 )
```

### Code/REPL

```python
import board

board.DISPLAY.refresh()
```

### Behavior

`sometimes` the busy line doesn't come back, ever. So the screen doesn't finish refreshing, so the system hangs.

### Description

I was unable to track completely what is happening how and would love insights from the experts on why this works.

Essentially: Sometimes the busy-line doesn't unlock, so the display.refresh() hangs and therefore hangs the complete CircuitPython.

### Additional information

I printline debuged Circuitpython on and found that flipping the switch of wait for busy in https://github.com/gdg-berlin-android/ZePython/blob/10bf2bafb474783c30970ffb7c2567357697b938/shared-module/epaperdisplay/EPaperDisplay.c#L210 does not seem to have any negative effects on the execution, but it doesn't halt the system, and more screen refreshes can happen.

I'm happy to understand more and why and how, therefore no PR for the change, because I'm lacking the complete picture on why this change works, what issues the busy line to never get triggered back again and what the implication of just ignoring it works. Help appreciated.

tannewt, 2024-04-30:
The busy line not being deasserted seems like a display problem, not a circuitpython one. What screen is on there? We do skip the busy pin for some boards because a display will also tell you how long it needs to refresh.

bablokb, 2024-05-01:
I have quite a few of these devices and some are running continuously and I never had any problems. So I assume this is an issue of the specific device.

What I did notice is that `time_to_refresh` never reaches zero but is stuck at 0.1. But maybe this is a misunderstanding of what this attribute does.

tannewt, 2024-05-01:
> What I did notice is that `time_to_refresh` never reaches zero but is stuck at 0.1. But maybe this is a misunderstanding of what this attribute does.

This may be a bug due to the rounding we added to prevent it from being too low.

mariobodemann, 2024-05-08:
Hey, 

Thanks for the replies so far, sorry for responding late. 

The board is a [badger2040](https://shop.pimoroni.com/products/badger-2040?variant=39752959885395), let me report back when I tried a different one.

bradanlane, 2024-05-26:
I just encountered to **time_to_refresh never reaching 0.0** bug.

I am handling paper refreshes through asyncio and track the time_to_refresh to know when it can be refreshed.

bablokb, 2024-05-27:
For the records: the issue is #9244 and creating a PR is on my todo-list.

dhalbert, 2025-07-29:
I believe this was fixed by #9299.

closed 2025-07-29 by dhalbert