adafruit/circuitpython #9215: i2ctarget documentation error filing and improvement request

XenonFlits, 2024-04-30:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.4 on 2024-04-16; Adafruit QT Py RP2040 with rp2040
```

### Code/REPL

```python
# n/a see Description
```

### Behavior

# n/a see Description

### Description

Hi, 

The sample on page https://docs.circuitpython.org/en/latest/shared-bindings/i2ctarget/index.html is kind of incomplete and insufficient. 
Also, there is an error in this page. 

First the error in formatting of the line "I2CTargetRequest.read() ack=False." at the bottom of the page. 
I believe it should be "I2CTargetRequest.read(ack=False)"

Second the sample code: It is of little practical value, to get one going.  It is even incomplete. 

For example: I try to connect an i2c master to a slave (i2c target with an address). The master is a microcontroller using a few lines in CircuitPython.  The slave is a microcontoller using CircuitPython too.  Both are on the same breadboard, same powerrail and i2c pins are interconnected, included is a pull-up resistor on each line. 

What I think one expects in the sample, is a scan to check the connection health of the target slave on i2c bus. Next logical step a read of a value over i2c of a slave register. 
What I think one expects next is a write of a value over i2c to a register in the slave. 
The current sample code does not provide this.

Even worse, in the sample code given, "if not r.is_read" is used with little explanation. Short names and non-related generic names make this sample extra difficult to read. (b, r, device). 
Longer and self-explanatory names like "busi2c" instead of device etc. 

Also, little is explained when and why ACK or NACK should be used on the bus. 

I know the i2c is quite complicated, I also know that to get it to work can be a headache. So two separate pieces of sample code that runs well between two microcontrollers using CircuitPython will be a good starting point for any coder interested in using CircuitPython "i2ctarget". 

Now Linux (undefined) is used as a master, which is of little help for a CircuitPython coder interested in in i2c communication between devices. 

Kind regards, 

Xf

dhalbert, 2024-04-30:
A Learn Guide with an example would be helpful.

ajmirsky, 2024-05-08:
@XenonFlits @dhalbert Happened to be recently working on understanding I2CTargets so I took what I learned and created some [additional documentation](https://github.com/mirskytech/circuitpython/commit/a9d800b3084c9341c6a09613b040d0a07f3efaf1?diff=split&w=0). If you have a moment, could you review and see if there are additional gaps that I haven't explained clearly?

Also, I believe there is a bug in 9.04 that prevents the full example (as well as the original example) from working:  during write-then-read transactions,  `is_restarted` is never set. I'll be filing that bug shortly and will post an update here when I do. In the meantime, you can work around this by [commenting out line 174-179](https://github.com/mirskytech/circuitpython/commit/a9d800b3084c9341c6a09613b040d0a07f3efaf1?diff=split&w=0#diff-6024355a071c69d8910ebc8da681deb43135269944d90f9d3f8a4ae98a1ec053R174-R179) but make sure you replace the assertion with a proper error check of the index.

ajmirsky, 2024-05-10:
FYI: https://github.com/adafruit/circuitpython/pull/9236

tannewt, 2024-05-15:
I think it is @ajmirsky. Thanks again!

closed 2024-05-15 by tannewt