adafruit/circuitpython #9220: mypy / pyright compatibility

bendavis78, 2024-05-03:
### CircuitPython version

```python
Installed in venv:

circuitpython-stubs==9.0.3
mypy==1.10.0
mypy-extensions==1.0.0
nodeenv==1.8.0
pyright==1.1.361
setuptools==69.5.1
typing_extensions==4.11.0
```

### Code/REPL

```python
n/a
```

### Behavior

n/a

### Description

My editor is configured to use pyright lsp and mypy for type checking. I'm working in a virtualenv with `mypy`, `pyright`, and `micropython-stubs` installed. I'm currently having the following issues:

### pyright
```
$ pyright code.py
/home/ben/Projects/personal/circuitpy/code.py
  /home/ben/Projects/personal/circuitpy/code.py:3:8 - warning: Import "board" could not be resolved from source (reportMissingModuleSource)
  /home/ben/Projects/personal/circuitpy/code.py:4:8 - warning: Import "digitalio" could not be resolved from source (reportMissingModuleSource)
  /home/ben/Projects/personal/circuitpy/code.py:8:36 - error: "LED" is not a known attribute of module "board" (reportAttributeAccessIssue)
1 error, 2 warnings, 0 informations
```
It's unclear what other packages my editor needs installed for mypy linting and code completion.

### mypy
```
$ mypy code.py
code.py:8: error: Module has no attribute "LED"  [attr-defined]
Found 1 error in 1 file (checked 1 source file)
```

With mypy, I'm unable to resolve any attributes for the `board` and `digitalio` module.

tannewt, 2024-05-03:
`digitalio` stubs are provided in https://pypi.org/project/circuitpython-stubs/ along with all of the other native modules. `board` is a special case because its contents change depending on the board. It takes extra setup: https://learn.adafruit.com/welcome-to-circuitpython/pycharm-and-circuitpython#device-specific-board-stubs-3168661

dhalbert, 2024-05-21:
Closing for now. Please reopen if you have further questions.

closed 2024-05-21 by dhalbert