adafruit/circuitpython #9223: Program not showing display but working with no errors in console

anonfaded, 2024-05-03:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.4 on 2024-04-16; Raspberry Pi Pico with rp2040
Board ID:raspberry_pi_pico
```

### Code/REPL

```python
# Raspberry Pi Pico

import board,busio
from time import sleep
from adafruit_st7735r import ST7735R
import displayio
import terminalio
from adafruit_display_text import label

mosi_pin = board.GP11
clk_pin = board.GP10
reset_pin = board.GP17
cs_pin = board.GP18
dc_pin = board.GP16

displayio.release_displays()

spi = busio.SPI(clock=clk_pin, MOSI=mosi_pin)

display_bus = displayio.FourWire(spi, command=dc_pin, chip_select=cs_pin, reset=reset_pin)

display = ST7735R(display_bus, width=128, height=160, bgr = True)

splash = displayio.Group()
display.root_groun = splash

color_bitmap = displayio.Bitmap(128, 160, 1)

color_palette = displayio.Palette(1)
color_palette[0] = 0x00FF00  # Bright Green

bg_sprite = displayio.TileGrid(color_bitmap, pixel_shader=color_palette, x=0, y=0)
splash.append(bg_sprite)

# Draw a smaller inner rectangle
inner_bitmap = displayio.Bitmap(118, 150, 1)
inner_palette = displayio.Palette(1)
inner_palette[0] = 0x000000  # Black
inner_sprite = displayio.TileGrid(inner_bitmap, pixel_shader=inner_palette, x=5, y=5)
splash.append(inner_sprite)

# Draw a label
text_group = displayio.Group(scale=1, x=11, y=24)
text = "Hello World!\n\nThis is a sample\ntext!\n\nEverything is \nworking fine."
text_area = label.Label(terminalio.FONT, text=text, color=0xFFFFFF)
text_group.append(text_area)  # Subgroup for text scaling
splash.append(text_group)

while True:
    pass
```

### Behavior

>>> %Run -c $EDITOR_CONTENT

### Description

- I am using raspberry pi pico 16mb flash, i followed [this ](https://youtu.be/KaGHxvVnKQ4?si=2DLmiU2r0WGSQ69d) tutorial from youtube and all steps were done right, pins etc everything is connected with the board and lcd module.
- I am using SPI TFT 1.8inch lcd module.
- The lcd is just white from start even before flash and now when i run that program to show some sample text on screen, it shows nothing and still its white, while in tutorial his same lcd was working fine. 
- The shell in Thonny IDE is also not showing any error and the program is running there, when i stop the screen is still white and i see no solution for this.

todbot, 2024-05-04:
You mispelled `root_group` when you're assigning the main group (you have `root_groun`). Not sure why that's not giving you an error.  
It's usually very instructive to add `print()` statements at various points in the code to see a) if the code is actually being run and b) how far it gets.  It may be freezing for some reason at a particular point and a couple of `print()` statements will help you narrow down where that is.

anonfaded, 2024-05-04:
I noticed that root_group typo but idk why it wasn't giving error for that, it happened during copy pasting it here.

But its still not working, i saved the code as `code.py` and i also used print statements, even till the end those print statements are working which can be seen in the ss below and lcd isn't showing any signs.
[image]

What if this is some driver issue or some issue with circuitpython? 
I'm sharing a screenshot of all the directories so if there is something missing you can check:
[image]

I bought the lcd and board new and i'm unable to identify the issue, for reference this is circuit i m using on breadboard:
| Display Pins | Board Pins      |
|--------------|-----------------|
| VCC          | 5V              |
| GND          | GND             |
| CS           | GP18 (CS)       |
| RESET        | GP17 (RESET)    |
| A0           | GP16 (DC)       |
| SDA          | GP11 (MOSI)     |
| SCK          | GP10 (SCLK)     |
| LED          | 3.3V            |

Code with print statements:
```python
# Raspberry Pi Pico

import board,busio
from time import sleep
from adafruit_st7735r import ST7735R
import displayio
import terminalio
from adafruit_display_text import label

mosi_pin = board.GP11
clk_pin = board.GP10
reset_pin = board.GP17
cs_pin = board.GP18
dc_pin = board.GP16

displayio.release_displays()

spi = busio.SPI(clock=clk_pin, MOSI=mosi_pin)

display_bus = displayio.FourWire(spi, command=dc_pin, chip_select=cs_pin, reset=reset_pin)

display = ST7735R(display_bus, width=128, height=160, bgr = True)

print("test1")
splash = displayio.Group()
display.root_group = splash
print("test2")

color_bitmap = displayio.Bitmap(128, 160, 1)

color_palette = displayio.Palette(1)
color_palette[0] = 0x00FF00  # Bright Green
print("test3")

bg_sprite = displayio.TileGrid(color_bitmap, pixel_shader=color_palette, x=0, y=0)
splash.append(bg_sprite)

# Draw a smaller inner rectangle
inner_bitmap = displayio.Bitmap(118, 150, 1)
inner_palette = displayio.Palette(1)
inner_palette[0] = 0x000000  # Black
inner_sprite = displayio.TileGrid(inner_bitmap, pixel_shader=inner_palette, x=5, y=5)
splash.append(inner_sprite)

# Draw a label
text_group = displayio.Group(scale=1, x=11, y=24)
text = "Hello World!\n\nThis is a sample\ntext!\n\nEverything is \nworking fine."
text_area = label.Label(terminalio.FONT, text=text, color=0xFFFFFF)
text_group.append(text_area)  # Subgroup for text scaling
splash.append(text_group)
print("test4")

while True:
    pass
```

todbot, 2024-05-04:
It all looks good from where I can see.  I would double check the wiring between display and Pico.  It's easy to get wires crossed, and since there's no communication back from the display, it's hard to know if you're actually talking to it. (For example, your code will work just fine with no display plugged in at all.  And the display will show white when powered but nothing hooked up to it)
Do you have a link to the exact display you bought? It may not actually have an ST7735 chip.

anonfaded, 2024-05-04:
This is the display lcd: https://click.daraz.pk/e/_CXh1bd
Also for exact wiring [this](https://youtu.be/KaGHxvVnKQ4?si=2DLmiU2r0WGSQ69d) tutorial can be checked as reference as i did exactly what he did and his display was also same as mine.

anonfaded, 2024-05-04:
> It all looks good from where I can see. I would double check the wiring between display and Pico. It's easy to get wires crossed, and since there's no communication back from the display, it's hard [...]

I just saw the description in it's ad and it mentioned the driver ic: ST7735s there is a s in the end, so i guess this js different, how can i make it work can you guide please.

todbot, 2024-05-04:
You should take this to the [#help-with-circuitpython Discord channel](https://discord.gg/XD45qb65). There are more people there that may have experience with this setup  This repo is for core CircuitPython stuff, you may also want to open up this issue in the [adafruit_st7735r driver](https://github.com/adafruit/Adafruit_CircuitPython_ST7735R/) repo as those folks may have experience with the "s" variant.

anonfaded, 2024-05-04:
I have opened an issue on the link you provided, I also messaged on Disocrd community so i will wait for their reply, thanks