adafruit/circuitpython #9234: pulseio.PulseIn maxlen is limited to 128 in esp32

tkomde, 2024-05-10:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.1-18-g781c577745-dirty on 2024-05-10; M5Stack AtomS3 with ESP32S3
```

### Code/REPL

```python
import board
import pulseio
import adafruit_irremote

pulsein = pulseio.PulseIn(board.D1, maxlen=1020, idle_state=True)
decoder = adafruit_irremote.GenericDecode()

while True:
    pulses = decoder.read_pulses(pulsein,max_pulse=91000)
    print(len(pulses))
```

### Behavior

prints "128"

### Description

It is possible that this is an intended specification, but I am reporting it.

I have a HITACHI air conditioner remote control data (848+6 data length) being read by pulseio.pulseIn. Regardless of the maxlen I gave it, it only read a maximum of 128 data.

I looked into the cause and found the following code in ports/espressif/common-hal/pulseio/PulseIn.c#103 .

```
self->raw_symbols_size = MIN(64, maxlen / 2 + 1) * sizeof(rmt_symbol_word_t);
````

When I delete the MIN(64, x) calculation as follows, the behavior differed depending on the CPU.
The ESP32S3(M5 AtomS3) and ESP32C6(M5 NanoC6) received all 854 data as desired, but the ESP32-PICO(M5 Matrix) still had 128 data. This seems to reproduce #7352.

```
self->raw_symbols_size = (maxlen / 2 + 1) * sizeof(rmt_symbol_word_t);
```

Is this MIN(64, x) decision required by some constraint? If almost esp32 chip can receive more data, I would like to receive it.

dhalbert, 2024-05-13:
`SOC_RMT_MEM_WORDS_PER_CHANNEL`, which is used in `PulseIn.c`, varies per processor. However, on the ESP32, it is `64`, and on ESP32-S3 and C6, it is `48`. Not sure what to make of that, given your test results above.

tannewt, 2024-05-13:
Its been a little while since I wrote this code. I think my intent was to limit the maxlen to the amount the RMT peripheral can buffer. I didn't want to assume PSRAM was accessible when PulseIn was running. (Writing the filesystem will disable it temporarily.)

I think some RMTs can use multiple channels worth of memory if needed. That could improve things.

Note each RMT word stores two "pulses" one active and one inactive. That's why 64 words give a length of 128.

tkomde, 2024-05-13:
@dhalbert 

Thanks for your comment.

This is a difficult area to understand for me, but I found an article that might be relevant. It seems that chips with the feature **SOC_RMT_SUPPORT_RX_PINGPONG** can handle more data than buffers. 

https://esp32.com/viewtopic.php?t=24484

CircuitFlyer, 2024-08-22:
Not to muddle up this topic but I also noticed a maxlen limit of 128 pulses on an ESP32-S3 QTPy no PSRAM.  My main issue was that PulseIn doesn't work at all to capture a set of pulses (<128) from a constant pulsed data stream.  Details [here](https://forums.adafruit.com/viewtopic.php?t=212959)

0wwafa, 2025-01-17:
So how to get continuously pulses and store them in an array without missing any?

dhalbert, 2025-03-27:
Raise an exception to note the limitation.

Sola85, 2025-05-24:
I also ran into this issue.

If I see correctly, `pulsio.PulseIn` was implemented for the espressif port before ESP-IDF version 5. Back then, DMA mode for the RMT peripheral was not implemented (see [here](https://docs.espressif.com/projects/esp-idf/en/v4.3.4/esp32/api-reference/peripherals/rmt.html?highlight=rmt)). But now in ESP-IDF Version >5, there is
[rmt_rx_channel_config_t::with_dma](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/peripherals/rmt.html#_CPPv4N23rmt_rx_channel_config_t8with_dmaE). If this is set to true, then  [rmt_rx_channel_config_t::mem_block_symbols](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/peripherals/rmt.html#_CPPv4N23rmt_rx_channel_config_t17mem_block_symbolsE) specifies the size of the internal DMA buffer.

> The receiver saves the incoming signals into its internal memory block or DMA buffer, in the format of [rmt_symbol_word_t](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/peripherals/rmt.html#_CPPv417rmt_symbol_word_t).

I have not tried this out yet, but maybe this could be a way of circumventing the limited internal RMT memory (?).

closed 2025-06-11 by dhalbert