adafruit/circuitpython #9235: Wemos C3 Mini should pre-set tx_power to 8.5

jepler, 2024-05-10:
### CircuitPython version

```python
main
```

### Code/REPL

```python
n/a
```

### Behavior

As reported by multiple users, with the power-on default tx_power, the board does not successfully connect to wifi (https://github.com/adafruit/circuitpython/issues/6540 https://github.com/adafruit/circuitpython/issues/9058#issuecomment-2104612719 and probably others) but does work if boot.py executes `wifi.radio.tx_power=8.5`.

### Additional information

The board init routine should set tx_power=8.5 by default, without needing lines in boot.py.

bill88t, 2024-06-20:
I did a bit more digging.
The reason `tx_power=8.5` is needed is not because of C3, but because of the type of antenna the board is using.

The common PCB antennas commonly on esp modules are what we most have.
However here and a bit of time I noticed **some** of my boards did not connect to one specific network.
Those boards all had 3D antennas (and are not C3's):
[image]

Then I got a MakerGo ESP32-C3 SuperTiny from aliexpress, that absolutely did not connect to any network.
That has a ceramic style antenna:
[image]

Setting the power on all of these, fixes that one network on the 3D antennas, and the supertiny entirely.
They all connect to every network I have. (Thinkpad, sbc intel wifi, phone hotspot, another esp32s3, an old router wifi - the one 3d antennas didn't originally connect to)

Furthermore, even the traditional pcb antennas seem to work fine with `tx_power=8.5`.
So it would probably be wise to set that as the default value.
It may not be optimal for range, but the alternative of going on every board that has a ceramic or 3d antenna and configuring it on `board_init()` doesn't sound all that better honestly.
It would ensure maximum compatibility.
For better range, the users are free to change that back during runtime.

dhalbert, 2024-06-20:
It appears from reading the documentation, though I am not sure of this, that `wifi.radio.tx_power=8.5` may be lowering the max power from its default. Would you be willing to do some testing to fetch and print out the power before and after the setting and see how its changed? It might get printed before USB is ready, so you could `ESP_LOGI()` it, or else turn off wifi workflow and do `import wifi` by hand to see the print.

bill88t, 2024-06-20:
@dhalbert The current default is 20.0, set by the "default wifi settings" from idf as far as I can tell.
So yes with 8.5 it's less than half power.
But that's what works on all the antenna types.
Compatibility vs range.

And besides, the tiny ceramic antenna did connect through closed doors to an AP 3 rooms away. It's honestly fine.

Also, as I already mentioned, for those who need the range, it's a settable value. They can crank it all the way.
However, the default should be what works on every board, every antenna type. Even if suboptimally.

It may be worth testing more values and antenna types though.

karlfl, 2024-06-21:
So, I took some time to perform some tests using the station example from the ESP-IDF. I discovered that my QTPY C3 would effectively connect up through power values of 18dBm (72 in the ESP API) but would never connect using the highest (and default) power of 20dBm (80 in ESP API).  Lower values would connect fairly well, but 18 seemed to connect within 1 or 2 tries every time.  

When I was digging into the [ESP WIFI API](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t), I discovered that you have to call this after wifi has started or it will error.  I then returned to the Circuit Python code (and reviewed the PR from @bill88t, which gave me good insight into where wifi start up was being called.  Thanks Bill!).  I then ran a few experiments within a cp build.  I discovered that placing the code in  `ports/espressif/common-hal/wifi/__init__.c` at the end of `common_hal_wifi_init()` seemed to prove successful.  Other places I tried this didn't seem to work, most likely because wifi wasn't started yet. There is probably a better place for this logic, but hopefully this gives you guys some insight into what I was able to get to work.  

Web Workflow comes up now without any boot.py modifications.  I plan to test this against some other wifi code (i.e. AP, web server, etc.) to see if all that works too.  

Here's a snippet showing my change...
```
void common_hal_wifi_init(bool user_initiated) {
    wifi_radio_obj_t *self = &common_hal_wifi_radio_obj;

    if (wifi_inited) {
        if (user_initiated && !wifi_user_initiated) {
            common_hal_wifi_radio_set_enabled(self, true); //<--  this starts radio, but may not be called at the right time
        }
        return;
    }
    wifi_inited = true;
    wifi_user_initiated = user_initiated;
    common_hal_wifi_radio_obj.base.type = &wifi_radio_type;

...

    // set station mode to avoid the default SoftAP
    common_hal_wifi_radio_start_station(self);
    // start wifi
    common_hal_wifi_radio_set_enabled(self, true);  // <-- radio has to be started before you can set the max power
    common_hal_wifi_radio_set_tx_power(self, 18); // <-- 18dBm anything higher won't be successful (QTPY C3)

}
```

bill88t, 2024-06-21:
It would be possible to detect error code 2 and recursively reduce  power and retry.
But that would take too long (in runtime) and take flash space.

bablokb, 2024-06-21:
@karlfl 
>  discovered that my QTPY C3 would effectively connect up through power values of 18dBm (72 in the ESP API) but would never connect using the highest (and default) power of 20dBm (80 in ESP API).

Nice insight. A few hours too late for me: I just replaced shortly before your post the onboard ceramic antenna on my QTPY C3 with a (very!) cheap external antenna and had a good and stable connection at once. Before it didn't even find all available networks when trying a scan.

dhalbert, 2024-06-21:
I wonder what the reason for the problem with the chip antennas is. I did some websearching and did not find any clear discussion. Maybe:
1. chip antenna has losses, is heating up, causing reduced performance
2. chip antenna is mismatched and higher reflected power due to higher SWR is affecting RF output stage

bablokb, 2024-06-21:
The chip antenna itself cannot be the problem. I have some Waveshare ESP32-S3 boards with an "C3"-ceramic antenna very similar to the one in the image above and they are working just fine. But the boards have a large ground-plane for the antenna (I guess 3-4 times the area of the antenna).

dhalbert, 2024-06-21:
I communicated with @ladyada about this. She is inclined to think it is mismatch and therefore reflected power, given that there is little room on some boards for an adequate layout.

bill88t, 2024-06-21:
I also had another find.
The other boards with ceramic antennas, like the Waveshare ESP32-S2 Pico (and probably the same S3 board) do not ground the ceramic antenna. There is a ground plain keepout area.
On the supertiny however, the antenna is grounded to the plain, intentionally.

What I don't understand now is why the 3D antennas connect to that one network with the reduced power.
They aren't grounded.

dhalbert, 2024-06-21:
The SuperTiny schematic, such as it is, is here: https://www.aliexpress.us/item/3256806297878595.html?gatewayAdapt=glo2usa4itemAdapt. 

Chip antennas are fed at one end and aren't meants to be grounded at the other end. But they need to be near a ground plane, at a particular orientation and distance: https://www.globalspec.com/learnmore/semiconductors/communication_ic/chip_antenna. Sometimes there isn't room for this or the designers are sloppy.

bill88t, 2024-06-21:
Ah? So in the schematic it isn't showing it as grounded?

But the physical thing is.
I desoldered the antenna.
The pad is connected to ground with a perfect 0Ω reading.

I would wanna just wanna say it's just a defective sample, but it's certainly not [the only one](https://community.home-assistant.io/t/esp32-c3-super-mini-wifi-signal-booster/657661).

And wait, the WeMos C3 Mini does keep (from the [image](https://www.wemos.cc/en/latest/_static/boards/c3_mini_v2.1.0_1_16x16.jpg) of it I can see on the web) 6.3mm from ground..

I give up trying to understand.

dhalbert, 2024-06-21:
The schematic doesn't show the "WIFI" antenna grounded. Which pad are you measuring? The one that's connected to UC8 and UR9?

Doesn't mean that there isn't some problem with the board.

bill88t, 2024-06-21:
> The schematic doesn't show the "WIFI" antenna grounded. Which pad are you measuring? The one that's connected to UC8 and UR9?

I was wrong. My horrible memory was remembering the WeMos ESP32 Lite (pcb ant).
The whole ant on that is shorted all the way to the usb-c shell.
That one works perfectly fine btw.

On the supertiny the antenna itself is shorted. It's not shorted to ground.

PR #9364 (2024-06-23): CIRCUITPY_DEFAULT_WIFI_TX_POWER
Alternative to #9357.
Fixes #9235.
closed 2024-06-23 by dhalbert