adafruit/circuitpython #9241: Memento WiFi breaks if espcamera (or PyCamera) object is created.

RetiredWizard, 2024-05-14:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.1 on 2024-04-19; Adafruit Camera with ESP32S3
```

### Code/REPL

```python
import io
import os
import board
import espcamera
import wifi
import adafruit_requests
import adafruit_connection_manager
from adafruit_io.adafruit_io import IO_HTTP, AdafruitIO_RequestError

if input('Create espcmera object? (Y/N): ').upper() == 'Y':
    cam = espcamera.Camera(
        data_pins=board.CAMERA_DATA,
        external_clock_pin=board.CAMERA_XCLK,
        pixel_clock_pin=board.CAMERA_PCLK,
        vsync_pin=board.CAMERA_VSYNC,
        href_pin=board.CAMERA_HREF,
        powerdown_pin=board.CAMERA_PWDN,
        reset_pin=None,
        i2c=board.I2C(),
        external_clock_frequency=20_000_000,
        pixel_format=espcamera.PixelFormat.JPEG,
        frame_size=espcamera.FrameSize.SVGA,
    )

print(f"Connecting to {os.getenv('CIRCUITPY_WIFI_SSID')}")
wifi.radio.connect(
    os.getenv("CIRCUITPY_WIFI_SSID"), os.getenv("CIRCUITPY_WIFI_PASSWORD")
)
print(f"Connected to {os.getenv('CIRCUITPY_WIFI_SSID')}!")

pool = adafruit_connection_manager.get_radio_socketpool(wifi.radio)
requests = adafruit_requests.Session(pool, adafruit_connection_manager.get_radio_ssl_context(wifi.radio))

# Initialize an Adafruit IO HTTP API object
aio_username = os.getenv("ADAFRUIT_AIO_USERNAME")
aio_key = os.getenv("ADAFRUIT_AIO_KEY")

io = IO_HTTP(os.getenv("ADAFRUIT_AIO_USERNAME"), os.getenv("ADAFRUIT_AIO_KEY"), requests)

# Get the 'camera' feed from Adafruit IO
feed_camera = io.get_feed("camera")
```

### Behavior

If you answer "Y" to the create espcamera object question the `io.get_feed("camera") instruction timeout/crashes as follows:
```
Create espcmera object? (Y/N): y
Connecting to twilightzone
Connected to twilightzone!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "espcam.py", line 41, in <module>
  File "adafruit_io/adafruit_io.py", line 748, in get_feed
  File "adafruit_io/adafruit_io.py", line 578, in _get
  File "adafruit_requests.py", line 591, in get
  File "adafruit_requests.py", line 525, in request
  File "adafruit_connection_manager.py", line 318, in get_socket
gaierror: (-2, 'Name or service not known')
```
If you answer "N" the script runs to completion without error

### Description

The same code runs without issue on AI Thinker ESP32-CAM board using the artifact from PR #9231. 

The issue appears to be either a problem specific to the Memento board or my board is faulty. I tried with and without the LED ring attached and got the same result. 

I found the issue when trying to adapt the code from the video doorbell learn guide so I suspect the issue was introduced in either a core update or an update to one of the networking libraries.

justmobilize, 2024-05-14:
This was way to familiar, found a similar issue: https://github.com/adafruit/circuitpython/issues/8907

dhalbert, 2024-05-14:
#8907 was supposedly fixed by #8913, so not sure if it is related.