adafruit/circuitpython #9275: Espressif ESP32-S3 DevKit-1 C boards moved NeoPixel pin on board revision

dhalbert, 2024-05-25:
Issue noted in https://forums.adafruit.com/viewtopic.php?t=211010 (wbeebe).

Espressif moved the RGB LED. Quoting from their [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html#hardware-revision-details):

>Both the initial and v1.1 versions of ESP32-S3-DevKitC-1 are available on the market. The main difference lies in the GPIO assignment for the RGB LED: the initial version uses GPIO48, whereas v1.1 uses GPIO38.

Creating new board defs seems like a bit of a waste. I wonder if we should alias the pins to, say, `board.NEOPIXEL_V10` and `board.NEOPIXEL_V11`, or maybe just omit the name.

bill88t, 2024-05-27:
For portable (across boards) code, this would be a bummer.
`board_id` is how code is supposed to tell what it's running on.

tannewt, 2024-05-29:
I'd prefer to have new boards for these. That way the status LED will work too. We can use `ccache` to speed up additional board builds.