adafruit/circuitpython #9276: ESPNow fails to initialize on ESP32 C6 DevKit C1 N8 board

felipejfc, 2024-05-25:
### CircuitPython version

```python
Tested both 9.1.0 beta 3 and 9.0.1
```

### Code/REPL

```python
]0;🐍Wi-Fi: off | REPL | 9.1.0-beta.3\
Adafruit CircuitPython 9.1.0-beta.3 on 2024-05-22; ESP32-C6-DevKitC-1-N8 with ESP32C6
>>> import os
>>> import espnow
>>> espnow.ESPNow()
```

### Behavior

Traceback (most recent call last):
]0;🐍Wi-Fi: No IP | REPL | 9.1.0-beta.3\  File "<stdin>", line 1, in <module>
espidf.IDFError: Generic Failure

### Description

Error happen right when trying to initialize espnow driver. board has no problem to use espnow on esp-idf native project

tannewt, 2024-05-29:
Hi @felipejfc, us Adafruit funded folks haven't done ESP-Now stuff. We're happy to help you debug this. The first step I do is with a debug build `DEBUG=1` and then see the UART output for ESP-IDF logging. We're available US work hours on Discord: https://adafru.it/discord in #circuitpython-dev

dhalbert, 2024-06-26:
This is a dupe of #9275. Espressif moved the NeoPixel pin from v1.0 to v1.1 of this board. A new board definition needs to be made.

dikei100, 2024-07-08:
Problem seems to be WiFi 6 mode on the ESP32C6 board. With the debug build I get this extended error message:

```
W (27220) wifi:This API is not supported in WIFI6 mode, please use esp_now_set_peer_rate_config() instead
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
espidf.IDFError: Generic Failure

```

Unfortunately, I haven't found a solution or workaround for this till now.

tannewt, 2024-08-29:
I think the solution is to migrate to `esp_now_set_peer_rate_config()`. I've asked [on the ESP forum](https://esp32.com/viewtopic.php?f=13&t=41600) what the values for the other parameters (phymode, ersu and dcm) should be so that it is backwards compatible.

tannewt, 2024-09-09:
Anyone know the parameters to use with the new API?