adafruit/circuitpython #9283: What is board_display_obj good for?

bablokb, 2024-05-30:
### CircuitPython version

```python
n.a.
```

### Code/REPL

```python
n.a.
```

### Behavior

n.a.

### Description

While porting CP to a new device, I discovered that many board.c files (I found 37) have this statement:

    fourwire_fourwire_obj_t board_display_obj;

Strange thing is that this variable is only declared, but never used within the complete codebase.

Is there some special magic creating code that actually uses this variable?

tannewt, 2024-05-30:
I think its leftovers from older code.

bablokb, 2024-05-31:
So I could create a PR removing these lines from all these files?

tannewt, 2024-05-31:
> So I could create a PR removing these lines from all these files?

Yes please!

PR #9294 (2024-06-02): remove definition of board_display_obj
This removes definitions of the otherwise unused variable `board_display_obj` in 36 `board.c` files.

Fixes #9283
closed 2024-06-02 by dhalbert