adafruit/circuitpython #9286: Add ability to enable byte clocking to bus_core

skerr92, 2024-05-31:
I have an ILI9488 SPI display that I've been working on a driver for. Unfortunately, I can't seem to get it to work with the standard bus device as it looks like the display driver is expecting each byte over SPI to be clocked individually rather than sending it all over one clock.

@bablokb has their own version of the ili9488 driver in which they use a custom build of circuitpython that employs the patch here [bus core patch](https://github.com/bablokb/circuitpython-examples/blob/master/waveshare-3.5/bus_core.patch)

I think it would be helpful to have an option to enable byte clocking if a display needs it while maintaining the current behavior by default.

skerr92, 2024-05-31:
I mean, either way there seems to be something not working right with any init sequence when using SPI with the ILI9488 devices when using SPI. I've tried with an init sequence derived from the data sheet, from TFT_eSPI (Arduino works), with similar sequences from adjacent displays. 

I still tend to think the issue might lay with how bus core is sending the spi data to the display. I've compared sending the same init sequence that works with Arduino, and it's not sending the same data. This is even comparing using ESP32, RP2040, and SAMD51.

tannewt, 2024-05-31:
This looks like normal SPI to me:
[image]

I think there is a bug on your side. I'd encourage you to use a logic analyzer to compare what CP does to your working Arduino example.

skerr92, 2024-05-31:
Do you have an idea of what it might look like to support RGB565? I'd be willing to take a look at the addition.

tannewt, 2024-05-31:
RGB565 is what we support currently. It is 16 bits.

You'll want something like RGB666 in 24 bits. I think the 3 bytes of 24 bits will be the hardest part because we usually only do power of twos 1, 2, 4 bits and 1, 2 and 4 bytes.

Most of the color conversion code is here: 
https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/ColorConverter.c

You'll likely need to change https://github.com/adafruit/circuitpython/blob/c47f013b9f7596fdaec67a71a891c96ec1f9df84/shared-module/displayio/TileGrid.c#L461 too for the 3 byte offsetting.

bablokb, 2024-05-31:
The ILI9488 also supports RGB565. That is what I am using. You will find that in the datasheet as well.

skerr92, 2024-05-31:
RGB565 is only supported in 8b and 16b data bus. SPI only supports RGB111 and RGB666.

skerr92, 2025-02-20:
@tannewt did the PR that was merged 2 months ago that tagged this issue resolve this issue? Specifically, did this target RGB666 SPI mode? If so, can we close this?