adafruit/circuitpython #9301: Can't run custom board build until large Build Boards CI job has run (giving cached submodules)

tyeth, 2024-06-04:
### CircuitPython version

```python
latest
```

### Code/REPL

```python
-
```

### Behavior

-

tyeth, 2024-06-04:
@dhalbert every seen anything like this? It fails with could not fetch 'user-fork' from the remote when talking about the esp-idf submodule, see this build log line:
https://github.com/tyeth/circuitpython/actions/runs/9358284335/job/25759793899#step:9:4428

tyeth, 2024-06-04:
Also Dan, my apologies again, as this one might have been better detected if I'd cleaned the cache before calling the script change good. 
Although we can switch the fork option to use the remote name origin, that feels hacky as heck so will see where the bug / issue is before making any radical changes.
Also worth noting the builds do work if the user doesn't cancel the large CI job when they first enable actions in their fork, or upon pushing any change. They also work in the adafruit repo as the large build boards CI job always runs and submodule assets are cached (except when someone 😇 cancels the job)

dhalbert, 2024-06-04:
I pinged you in discord if you want to discuss this now.

tyeth, 2024-06-04:
Just a note for future readers, and a reminder to self:
It was proposed the older ubuntu runners (22.04) may need git updating via PPA etc, so retest on latest runners.