adafruit/circuitpython #9302: EINPROGRESS when trying to connect to wifi the first time each day

dtorres-sf, 2024-06-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.3 on 2024-05-22; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
```

### Code/REPL

```python
# This is the failing line. I am use portalbase library to manage the feather esp32s3 and using the network class to do the connections and get the time from adafruit io
utc_time = self.portalbase.network.get_strftime("%Y-%m-%dT%H:%M:%SZ", location="UTC")
```

### Behavior

```Traceback (most recent call last):
  File "adafruit_connection_manager.py", line 271, in get_socket
OSError: [Errno 119] EINPROGRESS

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 282, in <module>
  File "code.py", line 245, in main
  File "code.py", line 130, in start_timer
  File "clockify.py", line 115, in start_clock
  File "adafruit_portalbase/network.py", line 237, in get_strftime
  File "adafruit_requests.py", line 597, in get
  File "adafruit_requests.py", line 535, in request
  File "adafruit_connection_manager.py", line 282, in get_socket
RuntimeError: Error connecting socket: [Errno 119] EINPROGRESS
```

### Description

The first time I try to connect to wifi each day I get this error. After that it works the rest of the day (even if power cycling in between).

tannewt, 2024-06-04:
How are you starting the device and connection each day? Is it a fresh power up?

dtorres-sf, 2024-06-04:
Good question. So I actually leave the device plugged into my docking station over USB. It is battery powered as well. So each morning I come in, hook up my laptop to the docking station which provides data and power to the device, but it is possible it was already powered on from the day before and running on battery but it would be in deep sleep mode.

dtorres-sf, 2024-06-06:
Last night I did not leave the USB connected to the docking station and this morning I did a full power cycle (disconnecting the battery) and started back up. It appeared to still hit the issue. I cannot be 100% certain since I wasn't connected to USB/serial to see the stack trace, but it hit the assert at the exact same time it usually does (and the app running has been unchanged for months and is pretty stable other than this issue).

tannewt, 2024-06-06:
Thanks for the info. That is weird! The simplest fix would be to catch the exception and retry.

To debug, we'd need the IDF logs that can be output over the UART.