adafruit/circuitpython #9304: ESP32-S3 crashes when a client subscribes to the Notify characteristic of the AccelerationService

tyeth, 2024-06-04:
Sorry @tannewt I've not managed to get a good debug output as it goes straight to safemode, but when a debug build I didn't see that it just stalled and serial eventually disconnected.

I was testing on the LilyGo T-Display-S3 (as the 4mb flash builds dont have BLEIO), and then eventually on the ESP32v2 Feather (8mb Flash / 2mb PSRAM).
Testing against the Web Bluetooth Dashboard https://adafruit.github.io/Adafruit_WebBluetooth_Dashboard/ with an SHT3x (sht30/31/35 temp+humidity sensor but not required) or BMP280 worked fine, but not with an accelerometer.

The code example linked below has been adapted to optionally ignore all non-present sensors, and accept one of 3 accelerometers (lsm9ds1 or lsm303 or lsm6ds33)

_Copied from discord [#](https://discord.com/channels/327254708534116352/334905778857050125/1246867992025894973):_
It's a reliable hard-fault 😄 Using an LSM303AGR (accelerometer) with the BLE Web Dashboard, hard fault as soon as it tries to start notifications from the AccelerationService, although I'm also seeing a console message claiming the web dashboard is attempting to update the measurement period which happens before notify start (I set it to 500 ms period instead of 100 in code so shouldn't be required, but maybe not part of broadcast info so writes desired value rather than read then write if incorrect). 
Modified example to skip non-available sensors and have 3 Accelerometer options, https://github.com/tyeth/Adafruit_CircuitPython_BLE/blob/web-bluetooth-dashboard/examples/ble_bluefruit_web_dashboard.py, I'll try to capture more details before filing an issue.
BLE Web Dashboard: https://kind-decisive-paw.glitch.me/ 

Original learn guide: https://learn.adafruit.com/bluefruit-dashboard-web-bluetooth-chrome

tannewt, 2024-06-13:
It'd be easier to debug this if we had CircuitPython client code that trigger the issue.

tyeth, 2024-06-27:
@dhalbert you'll be relieved to hear your work on #9374 has fixed it. Thank you both for tackling the BLE stuff.
I initially thought ah-almost, as the T-Display S3 just didn't crash this time but no accelerometer data appeared in browser, but reading the console I saw it moan about a missing service (battery service it turned out), which the T-Display won't/doesn't have.
Tried again on feather s3 reverse tft, but not enough space for BLE, so ran a custom debug build using 
`CIRCUITPY_BLEIO=1 FLASH_SIZE_SDKCONFIG=esp-idf-config/sdkconfig-flash-4MB-no-ota.defaults` for build flags (inspired by memento def) and hey presto I see acceleration service data (and battery with a ? as none attached which I'd never thought about really)
[image]

dhalbert, 2024-06-27:
@tyeth Fantastic, thank you for testing!

dhalbert, 2024-06-27:
Fixed by #9374.

closed 2024-06-27 by dhalbert