adafruit/circuitpython #9305: Settable microcontroller.cpu.frequency on espressif builds

Sola85, 2024-06-04:
Currently only "RP2040 boards and some i.MX boards" support changing the CPU frequency. In particular, none of the esp32 boards do, despite the hardware being capable of this in principle.

Is this not implemented simply due to lack of significance, or was there a deliberate decision against this? (E.g. was this tried and determined to result in unstable USB/Peripherals/... ?)

Looking through previous discussions on this topic (e.g. https://github.com/adafruit/circuitpython/issues/2284, https://github.com/adafruit/circuitpython/issues/4176, https://github.com/adafruit/circuitpython/issues/4339 and https://github.com/adafruit/circuitpython/pull/6279), I find no indication of this being attempted for esp32 builds, which makes me guess that simply no one got around to implementing this.

A possible implementation could follow that of micropython [here](https://github.com/micropython/micropython/blob/932f76c6ba64c5a3e68de3324556d9979f09303b/ports/esp32/modmachine.c#L101), or the Arduino ESP32 core [here](https://github.com/espressif/arduino-esp32/blob/c16a325439fe5a8a3160701d9601634155bb1443/cores/esp32/esp32-hal-cpu.c#L176). Would a PR implementing either of the two options have a chance of being accepted?

tannewt, 2024-06-04:
Yup! We'd totally accept it. We just haven't had anyone implement it.

dhalbert, 2024-06-04:
Any precomputed values or peripheral settings that depend on the current clock rate would need to be changed, unless we specifically said "that's your problem".

tannewt, 2024-06-04:
We do [warn in the docs](https://docs.circuitpython.org/en/latest/shared-bindings/microcontroller/index.html#microcontroller.Processor.frequency) already:
[image]

closed 2024-07-08 by tannewt