adafruit/circuitpython #9306: Raw HID [get_last_received_report()] broken on 9.1.0-beta.3 (Pi Pico rp2040)

dtcooper, 2024-06-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.3 on 2024-05-22; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
#### boot.py ####

# rawhid_boot.py --
# any changes here only reflected after board reset

import usb_hid

REPORT_COUNT = 63  # size of report in bytes

CUSTOM_REPORT_DESCRIPTOR = bytes((
    0x06, 0x00, 0xff,  # Usage page (Vendor Defined Page1)
    0x09, 0x01,        # Usage (Vendor Page 1)
    0xA1, 0x01,        # Collection (Application)

    0x85, 0x01,        # Report ID (1)
    0x09, 0x00,        # Usage Page (Undefined)
    0x15, 0x00,        # Logical Minimum (0)
    0x26, 0xFF, 0x00,  # Logical Maximum (255)
    0x75, 0x08,        # Report Size (8 bits)
    0x95, REPORT_COUNT, # Report Count (64 fields)
    0x82, 0x02, 0x01,  # Input (Data,Var,Abs,Buf)

    0x85, 0x02,        # Report ID (2)
    0x09, 0x00,        # Usage (Undefined)
    0x09, 0x00,        # Usage Page (Undefined)
    0x15, 0x00,        # Logical Minimum (0)
    0x26, 0xFF, 0x00,  # Logical Maximum (255)
    0x75, 0x08,        # Report Size (8 bits)
    0x95, REPORT_COUNT, # Report Count (64 fields)
    0x92, 0x02, 0x01,  # Output (Data,Var,Abs,Buf)

    0xC0,        # End Collection
))

raw_hid = usb_hid.Device(
    report_descriptor=CUSTOM_REPORT_DESCRIPTOR,
    usage_page=0xff00,         # Vendor defined
    usage=0x01,                # Vendor page 1
    report_ids=(1, 2),
    in_report_lengths=(REPORT_COUNT,0),
    out_report_lengths=(0, REPORT_COUNT),
)

usb_hid.enable( (raw_hid,) )

#### code.py ####
# rawhid_code.py  -- copy to CIRCUITPY as "code.py"
# don't forget to install rawhid_boot.py as "boot.py" and press reset
# works with report IDs up to 63 byte report count
# test with hidapitester like:
# ./hidapitester --usagePage 0xff00 --usage 1 --open -l 64 --send-output 2,3,4,5 --timeout 1000 --read-input 1
# adapted from code presented here:
# https://github.com/libusb/hidapi/issues/478

import time
import usb_hid
import adafruit_hid

raw_hid = adafruit_hid.find_device(usb_hid.devices, usage_page=0xff00, usage=0x01)

print("raw_hid: %04x %04x" % (raw_hid.usage_page, raw_hid.usage) )

while True:
    out_report = raw_hid.get_last_received_report(2) # out from computer
    if out_report:
        print("len:",len(out_report),["%02x" % x for x in out_report])
        time.sleep(0.5)
        print("sending copy on reportid 1")
        in_report = bytearray(out_report)  # copy in case we want to modify
        raw_hid.send_report(in_report, 1);  # in to computer
```

### Behavior

No output observed, whereas in 9.1.0-beta.2 the above example works perfectly

### Description

Given @todbot's example of creating a vendor-defined raw HID to communicate over USB &mdash; [gist found here](https://gist.github.com/todbot/6c39e9f2e9719643e5be8f1c82cf9f79) and [further discussion in GitHub issues found here](https://github.com/adafruit/circuitpython/issues/5529), I observe communication from my computer to Circuitpython broken as of **9.1.0-beta.3**. Note: it works find in **9.1.0-beta.2**.

### Additional information

Using **9.1.0-beta.2** is a workaround for now, but hope to help prevent this regression in **9.1.x**. Thanks!

dhalbert, 2024-06-04:
@hathach This is of interest to you. We updated TinyUSB in 9.1.0-beta.3.

dhalbert, 2024-06-10:
I believe this is due to https://github.com/hathach/tinyusb/pull/2253. I'm checking to see how the arguments to `tud_hid_set_report_cb()` have changed.

dtcooper, 2024-06-10:
Great, thanks for looking into this.

dtcooper, 2024-06-10:
#9322 appears to fix the issue. See pull request for more info. :)

closed 2024-06-11 by tannewt

dtcooper, 2024-06-11:
👏 thanks again!

PR #9322 (2024-06-11): Update `tud_hid_set_report_cb()` report handling
- Fixes #9306.

https://github.com/hathach/tinyusb/pull/2253 changed the `report_type` value that is passed to `tud_hid_set_report_cb()`. Handle this change.

@dtcooper Could you test for your raw HID case? I tested this with keyboard and keyboard LED code, not raw HID. Thanks. The checks Summary page will include build artifacts.