adafruit/circuitpython #9312: QT PY RP2040 Pins A0-A3 not capable of enabling internal pull up/down resistors

juansulca, 2024-06-06:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.5 on 2024-05-22; Adafruit QT Py RP2040 with rp2040
```

### Code/REPL

```python
import board
from digitalio import DigitalInOut, Direction, Pull

btn_0 = DigitalInOut(board.D0)
btn_0.direction = Direction.INPUT
btn_0.pull = Pull.UP

btn_1 = DigitalInOut(board.D4)
btn_1.direction = Direction.INPUT
btn_1.pull = Pull.UP
```

### Behavior

When I measure the voltage between GND and pin D4, it is set to +3.3v as expected but pin D0 is set to 0. This same issue is present with pins A0/D0, A1/D1, A2/D2, A3/D3. I don't know if this is intended behavior or if I am misusing the pins.

johnnohj, 2024-06-06:
Wasn't able to replicate this. Does it function correctly if you conclude your code with:

`while True:
	pass` 

Or if you copy the code into the REPL?

juansulca, 2024-06-20:
Sorry for the delay, I tried using the REPL and reflashed my board with the latest version of circuit python and the problem went away. It is working fine now. Thanks a lot for the reply.

speedbird620, 2025-01-18:
I got the same issue with version 9.2.3:

`import digitalio, board, time
from adafruit_hid.keyboard import Keyboard

pin0 = digitalio.DigitalInOut(board.D0)
pin0.direction = digitalio.Direction.INPUT
pin0.pull = digitalio.Pull.UP

pin4 = digitalio.DigitalInOut(board.D4)
pin4.direction = digitalio.Direction.INPUT
pin4.pull = digitalio.Pull.UP

time.sleep(0.5)

print("Pin 0: " + str(pin0.value) + ", pin 4: " + str(pin4.value))`

The output is:

`Pin 0: False, pin 4:True`

@juansulca, which version did you use to get it going?

dhalbert, 2025-01-18:
@speedbird620 Please add
```
while True:
    pass
```
to the end of your program, or else try it line by line in the REPL.

speedbird620, 2025-01-18:
@dhalbert, I tried. No joy.

Please enlighten me, what will "pass" change in this case?

speedbird620, 2025-01-18:
Sorry. My bad. I connected the QT PY the wrong way. 

It works now.

dhalbert, 2025-01-18:
I am not seeing this problem with QT Py RP2040 running 9.2.3.
```
Adafruit CircuitPython 9.2.3 on 2025-01-17; Adafruit QT Py RP2040 with rp2040
>>> import board
>>> from digitalio import DigitalInOut, Direction, Pull
>>> 
>>> pin0 = DigitalInOut(board.D0)
>>> pin0.direction = Direction.INPUT
>>> pin0.pull = Pull.UP
>>> 
>>> pin4 = DigitalInOut(board.D4)
>>> pin4.direction = Direction.INPUT
>>> pin4.pull = Pull.UP
```
when measured with a voltmeter I am seeing 3.3V on both D0 and SCL.
D4 is the same as the pin marked `SCL` on the board.

What is the exact name of the UF2 file you used?
What are the contents of `boot_out.txt`?
Could you take a picture of the top of the board and point to the pins you are using?

dhalbert, 2025-01-18:
Glad it's working! Could you describe what "the wrong way" was in this case? It sounds like it might have happened to more than one person.

speedbird620, 2025-01-18:
It was an error on the PCB I designed. The button was shorting the pin to the ground whether it was pressed or not.  

A simple case of Googling the symptom before performing the due diligence check. Game, set, match. 

Anyways, thanks for the input.