adafruit/circuitpython #9320: `microcontroller.cpu.temperature` watchdog safemode on ESP32-S3 when wifi connected

bill88t, 2024-06-09:
### CircuitPython version

```python
Current master
CONFIG_FREERTOS_UNICORE=y
adc_paranoia (the name of the adc multi-sampling branch)
and pretty much every other build I tested
```

### Code/REPL

```python
# Assuming wifi connected via settings.toml
import microcontroller
while True:
    a = microcontroller.cpu.temperature
# `a =` is there to not slow down with printing to serial
# It can take a lot longer than the ADC .value bug to proc.
```

### Behavior

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.

dhalbert, 2024-06-10:
I think this is another manifestation of #9291, since the temperature reading probably uses ADC functionality.

dhalbert, 2024-06-11:
Closing in favor of #9291.

closed 2024-06-11 by dhalbert

bill88t, 2024-08-18:
Fixed with #9526.
#9291 is still not fixed.