adafruit/circuitpython #9323: wifi.radio.ap_info was never implemented on the Pico W

justmobilize, 2024-06-10:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.4 on 2024-04-16; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import wifi
hasattr(wifi.radio, "ap_info")
```

### Behavior

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError:
```

### Description

`wifi.radio.ap_info` was never imlemented on the [Pico W](https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/wifi/Radio.c#L341C1-L343C2)

RnoldR, 2025-03-25:
Any change this gets done? The examples assume wifi.radio.ap_info exists. Maybe just a working example for the Pico (and any other board that needs this attribute).

deelowe, 2025-07-07:
Not sure if this is what's preventing implementation, but this thread found a solution for getting the rssi for a connected AP:

> Hmm, seems the Pico SDK/cyw43 driver has the Infineon numbers doubled / shifted left?
> 
> 0x19 * 2 = 0x32
> 0x1d * 2 = 0x3a
> 
> 
> May want to try if using 127 * 2 = 254 (0xfe) works as command to get RSSI?

Source: https://forums.raspberrypi.com/viewtopic.php?p=2047463#p2047467

closed 2026-10-25 by dhalbert