adafruit/circuitpython #9327: M5STACK FIRE failed boot

reltkaine, 2024-06-12:
Tried uploading firmware 9.0.4 9.1.0 and 8.2.10....but seems failed at boot....i used thonny....

jepler, 2024-06-16:
@CDarius it looks like you contributed this board in #7113 and #7144, are you able to check and confirm whether it's working for you?

reltkaine, 2024-06-17:
Tried pressing btn A (9.0.5 9.1.0 and 8.2.10) no avail and notice that the power button seems not functioning ON OFF function....
Even trying to turn ON/OFF the display has a bit of background light......

CDarius, 2024-06-17:
I'm away from home for a couple of weeks, I can't make any test at the moment

CDarius, 2024-07-05:
@reltkaine I made a simple test: I installed UIFlow on the M5Stack Fire and then I have reinstalled CircuitPython using the webinstaller on circuitpython.org and it works

reltkaine, 2024-07-16:
M5Stack Fire v2.6 not working me...

reltkaine, 2026-01-06:
I check my m5fire and the circuitpython port files and I found out that my m5fire is really the earliest version that uses a PSRAM of 4MB rather than 8MB... I'm hoping if you can rebuilt a 4MB PSRAM Binary Version......

dhalbert, 2026-01-07:
It is a continuous source of overhead on every build to add a board, and we also have to add it to circuitpython.org. We don't mind adding boards, but if the number of users would be vanishingly small, I personally would prefer not to.

Learning how to do your own builds requires some work, but then you could make new builds for this board as you wish. https://learn.adafruit.com/building-circuitpython.