adafruit/circuitpython #9330: ship mpy-cross similar to micropython-uncrustify

jepler, 2024-06-12:
Recently micropython dealt with the difficulty of getting the right uncrustify version by creating a micropython-uncrustify package. This looks like it'll install on windows, linux & mac.

we could do mpy-cross the "same way", probably having an mpy-cross-<circuitpython version> package so that different mpy-cross binaries are co-installable.

then, use this in circuitpython-build-tools, get rid of all the hard coding of download paths in various tools, etc.

dhalbert, 2024-06-12:
GMTA. This was just added to pypi: https://pypi.org/project/mpy-cross-multi/

jepler, 2024-06-13:
I wonder if we could ask mpy-cross-multi to support circuitpython :wink:

jepler, 2024-06-13:
mpy-cross-multi seems to be built on various mpy-cross pypi packages like https://pypi.org/project/mpy-cross-v5/ which are not on github (but there is a source distribution on pypi)