adafruit/circuitpython #9333: Setting the name property on an adafruit_ble.BLERadio instance doesn't work on ESP32-S3

ashirvadz, 2024-06-14:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.3 on 2024-05-22; FeatherS3 with ESP32S3
```

### Code/REPL

```python
import time
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService
from adafruit_ble.uuid import StandardUUID
from adafruit_ble.services import Service
from adafruit_ble.characteristics import Characteristic
from adafruit_ble.characteristics.string import StringCharacteristic

# Define a custom service and characteristic
class CustomService(Service):
    uuid = StandardUUID(0x181C)  # Custom service UUID
    custom_characteristic = StringCharacteristic(
        uuid=StandardUUID(0x2A58),  # Custom characteristic UUID
        properties=(Characteristic.READ | Characteristic.WRITE),
        initial_value=b"Hello BLE!",
    )

# Initialize the BLE radio
ble = BLERadio()
ble.name = "MyBLEDevice"

# Create the custom service instance
custom_service = CustomService()

# Create the advertisement
advertisement = ProvideServicesAdvertisement(custom_service)
advertisement.complete_name = "MyBLEDeviceAdv"

# Start advertising
ble.start_advertising(advertisement)
print("Advertising Custom Service...")

while not ble.connected:
    time.sleep(0.1)

print("Device Connected")

while ble.connected:
    # Perform tasks while connected
    pass

print("Device Disconnected")

# Stop advertising when disconnected
ble.stop_advertising()
```

### Behavior

The device name shows as "CIRCUITPY", and the name property show as "Local Name" on a BLE scanner app such as Bluefruit Connect:
[image]

dhalbert, 2024-06-25:
This is often because the host reporting the name has cached a previous name, and doesn't bother to change the name. I've seen this a lot on iOS. You can try turning Bluetooth off and back on, on the host, and "forgetting" the device. Does that make a difference?

ashirvadz, 2024-06-27:
It doesn't seem to be a cache issue. I've followed the steps you suggested, and they did not make a difference.

dhalbert, 2024-06-27:
Are you testing this on iOS? I want to replicate your testing conditions. Thanks.

ashirvadz, 2024-06-27:
Yes, I am testing it on iOS.

dhalbert, 2024-07-02:
I tried your test program with the tip of `main`, as of this comment. I saw the device name as `MyBLEDevice` in iOS nRFConnect and in Adafruit Bluetooth Connect. I then edited the name to be `MyBLEDevice2`. If I simply restart the program without a hard reset, or if I do a hard reset, both monitoring programs showed the new name. I changed it back, and then to `MyBLEDevice3`, and the new name showed up. Could you try again with the absolute newest build from https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin ? Thanks.

ashirvadz, 2024-07-03:
I have flasehd the latest build, and the name change works on nRFConnect. However on Adafruit Bluetooth Connect it still shows the device name as CIRCUITPY(id), with the Local Name property set as the new name. I tried restarting the board and the phone, without luck. Anything else I should try?

[image]

[image]

dhalbert, 2024-07-04:
Since nRFConnect is showing the right name, but Bluefruit Connect is not, this is not a problem with what the board is transmitting. It is some kind of caching problem. So I'm moving this to be a support issue.

Is the app updated to the latest version?
Which iOS device model and which iOS version are you using?

ashirvadz, 2024-07-04:
You are right, it seems to be a caching problem. I installed Adafruit Bluefruit Connection on another device, and it showed the correct name. Thank you!

daniel-alsen, 2025-07-11:
I had similar issues and looked into this a bit more in detail. It's working for me if I set a short device name on the BLERadio.name property only and use that name in scanning (I use the service UUID in addition).

Some findings:

- A too big BLE advertising packet creates issues (AI says 31 bytes max)

If I put in a too long name, or populate both the name and local name in a way that the total packet size get's too big, I run into different types of issues depending on the scanning software (e.g. iOS, Light Blue, nRF, etc.).

On iOS, the device was not visible at all in some cases. In other cases it showed up as "CIRCUITPYXXX", which is probably an old cached name although I don't think I have connected before naming the device, which is strange

- CircuitPython: BLERadio.name is the property to use for a short name. Advertisement.complete_name doesn't seem to have any effect, and I ran into issues when I populated both BLERadio.name and Advertisement.short_name. Perhaps this is advertising packet size related?

- The "name" seen in the scanning software seems to be the scanning softwares best guess of what the name can be. It can be populated by either the Complete Local Name (Advertising packet type 0x09), the Shortened Local Name (Advertising packet type 0x08), the Device Name received after connecting to the device (GATT Characteristic 0x2A00) or some combination of those with IDs, or shortened versions

- The device name when last connected seems to often be used in the scanning software cache, if there isn't a problem with the advertising paket size

>> CircuitPython - Version 10 - Alpha 7

(Docs)
Class: BLERadio
Property: name - The name for this device. Used in advertisements and as the Device Name in the Generic Access Service, available to a connected peer.

Class: Advertisement
Property: short_name - Short local device name (shortened to fit).
Property: complete_name - Complete local device name.

dhalbert, 2025-07-29:
Closing this because I think it is resolved. It might be nice to write up the caching discoveries as a Playground note.