adafruit/circuitpython #9334: storage.remount disable_concurrent_write_protection

mikeysklar, 2024-06-14:
### CircuitPython version

```python
CircuitPython 9.0.5
Feather RP2040
```

### Code/REPL

```python
import storage

# Remount the filesystem as writable with concurrent write protection disabled
storage.remount("/", readonly=False, disable_concurrent_write_protection=True)
```

### Behavior

Still seeing drive mounted as read-only when laptop is rebooted, but Feather is powered from USB hub. 

### Description

In this case the desired behavior is to override the write_protection so development work can be resumed when the computer USB connection comes back. 

MacOS Ventura (13.6.3)
MacBook Air M2

### Additional information

Based on Forum Thread from @blakebr who shutdown his Windows desktop each night and would like to resume work on his USB hub powered controllers in the AM. His testing was on Windows and mine was on MacOS.

simple cpu temp code running as code.py

```
import time
import microcontroller

while True:
    print(microcontroller.cpu.temperature)
    time.sleep(5)
```

dhalbert, 2024-06-14:
Is the remount in boot.py? Could you check boot_out.txt and see that no errors were written there?

dhalbert, 2024-06-14:
Also please try 9.1.0-beta.3.

mikeysklar, 2024-06-14:
```
sklarm@grazie CIRCUITPY % cat boot_out.txt
Adafruit CircuitPython 9.0.5 on 2024-05-22; Adafruit Feather RP2040 with rp2040
Board ID:adafruit_feather_rp2040
UID:4545373038057E2C
boot.py output:
```

I will try 9.1.0-beta.3.

RetiredWizard, 2024-06-14:
Can you post boot_out.txt, code.py and boot.py contents from the device?

mikeysklar, 2024-06-14:
boot_out.txt

```
Adafruit CircuitPython 9.1.0-beta.3 on 2024-05-22; Adafruit Feather RP2040 with rp2040
Board ID:adafruit_feather_rp2040
UID:4545373038057E2C
boot.py output:
```

boot.py
```
import storage

# Remount the filesystem as writable with concurrent write protection disabled
storage.remount("/", readonly=False, disable_concurrent_write_protection=True)
```

code.py
```
# SPDX-FileCopyrightText: 2021 Kattni Rembor for Adafruit Industries
# SPDX-License-Identifier: MIT
"""CircuitPython CPU temperature example in Celsius"""
import time
import microcontroller

while True:
    print(microcontroller.cpu.temperature)
    time.sleep(5)
```

[feather-rp2040-files.zip](https://github.com/user-attachments/files/15841246/feather-rp2040-files.zip)

mikeysklar, 2024-06-14:
Yes, both @b-blake and I are using powered USB hubs so the controller continues to run when the attached PC goes away. 

Everything works as expected when the Feather RP2040 is reset. Things don't go well when the PC goes away and comes back. It is either read-only mount or no mount of the CIRCUITPY drive.

b-blake, 2024-06-17:
@dhalbert;
> Is the remount in boot.py? 
> Could you check boot_out.txt and see that no errors were written there?
There is no remount in boot.py
There are no errors in boot_out.txt

This is the output of CircUp.exe this morning from my Windows 10 PC after having it off all night.  It was the first thing run at boot-up.  If you want specifics about any device. please ask.

` Volume in drive C is Windows SSD
 Volume Serial Number is B984-5BB6

 Directory of C:\Users\Bruce\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts

06/07/2024  06:54 PM    <DIR>          .
06/07/2024  06:54 PM    <DIR>          ..
06/07/2024  06:54 PM           108,455 circup.exe
01/26/2023  06:41 PM             1,817 circup.txt
11/13/2022  06:11 PM           107,948 findimports.exe
11/13/2022  06:11 PM           107,982 normalizer.exe
03/14/2024  08:16 PM           108,471 pip.exe
03/14/2024  08:16 PM           108,471 pip3.10.exe
03/14/2024  08:16 PM           108,471 pip3.exe
11/13/2023  09:33 PM           108,459 pysemver.exe
03/07/2024  03:22 PM             8,820 temp.txt
               9 File(s)        768,894 bytes
               2 Dir(s)  538,066,677,760 bytes free

CircUp, A CircuitPython module updater. Version 2.0.2

Do you want to upgrade? [Y/n]:

Waiting for 14 seconds, press a key to continue ...
####################################### H:
circup --path H: update --all
Found device at H:, running CircuitPython 9.1.0-beta.3.
Downloading latest bundles for adafruit/Adafruit_CircuitPython_Bundle (20240617).
py:
Extracting:  [####################################]  100%
8.x-mpy:
Extracting:  [####################################]  100%
9.x-mpy:
Extracting:  [####################################]  100%

OK

Downloading latest bundles for adafruit/CircuitPython_Community_Bundle (20240617).
py:
Extracting:  [####################################]  100%
8.x-mpy:
Extracting:  [####################################]  100%
9.x-mpy:
Extracting:  [####################################]  100%

OK

Found 3 module[s] needing update.
Something went wrong, [WinError 19] The media is write protected: 'H:lib\\adafruit_logging.mpy' (check the logs)
Something went wrong, [WinError 19] The media is write protected: 'H:lib\\adafruit_displayio_ssd1306.mpy' (check the logs)
Something went wrong, [WinError 183] Cannot create a file when that file already exists: 'H:lib\\asyncio\\' (check the logs)

Waiting for 0 seconds, press a key to continue ...
####################################### I:
circup --path I: update --all
Usage: circup [OPTIONS] COMMAND [ARGS]...
Try 'circup --help' for help.

Error: Invalid value for '--path': Directory 'I:' does not exist.

Waiting for 0 seconds, press a key to continue ...
####################################### J:
[... 105 lines cut ...]
Usage: circup [OPTIONS] COMMAND [ARGS]...
Try 'circup --help' for help.

Error: Invalid value for '--path': Directory 'V:' does not exist.

Waiting for 0 seconds, press a key to continue ...
####################################### W:
circup --path W: update --all
Found device at W:, running CircuitPython 9.0.5.
None of the module[s] found on the device need an update.

Waiting for 0 seconds, press a key to continue ...
####################################### X:
circup --path X: update --all
Found device at X:, running CircuitPython 9.1.0-beta.3.
None of the module[s] found on the device need an update.

Waiting for 0 seconds, press a key to continue ...

 Volume in drive C is Windows SSD
 Volume Serial Number is B984-5BB6

 Directory of C:\Users\Bruce\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts

Press any key to continue . . .`