adafruit/circuitpython #9336: is /cp/serial/ working?

gitcnd, 2024-06-15:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.1-24-g24f2e4f994 on 2024-05-18; S2Mini with ESP32S2-S2FN4R2
and
Adafruit CircuitPython 9.1.0-beta.3-28-g03e42a8c0c-dirty on 2024-06-15; AButton V3 with ESP32
and
Adafruit CircuitPython 9.0.5 on 2024-05-22; S2Mini with ESP32S2-S2FN4R2
```

### Code/REPL

```python
Connecting to my cpy boards delivers the "welcome!" screen.  They include a "serial terminal" [sic] link which is a websocket (not serial) connection to the REPL as far as I can tell.

None of these work for me
```

### Behavior

clicking "send" does nothing visible in my browser.
if I connect over serial (real serial - hardware RS232) I do notice that clicking the send button above locks up my serial console though, so something seems to be trying to work.

I've got a full working build system installed, so happy to help troubleshooot and fix this, if anyone wants to give me some pointers.

dhalbert, 2024-06-16:
This is working for me, on Matrix Portal S3, in both the latest Firefox and Chrome, on Linux. I can click `Send` or press enter.
[image]

tannewt, 2024-06-17:
What browser are you using? Are there errors in the browser's developer tools?

gitcnd, 2024-06-18:
Ahh - you use the IP address, right?

Try again using the DNS name:

http://cpy-eaf90c.local/cp/serial/

I notice it works with the IP address, but not with the name.

I'm using Chrome on Windows; windows does properly resolve mdns so the web page at http://cpy-eaf90c.local/ loads and works fine, but not the /cp/serial/ part.

ymmv with ubuntu; I never figured out how to get that understanding mdns (at least not in WSL anyhow)

gitcnd, 2024-06-18:
update - I notice it intermittently returns some parts of what got sent, after a multi-minute delay, if you leave the browser open for a long time.

dhalbert, 2024-06-18:
It works fine for me with both IP and cpy-nnnn.local names. I'm assuming your browsers are up to date. I've tested with all three operating systems. My LAN is running on a Verizon FIOS router.

Have you set the DNS server for your OS's to be your router (e.g., 192.168.x.x) or have you set DNS to directly go somewhere else, such as 8.8.8.8 (Google)?

Are you running a code.py that is doing something? If so, try removing code.py and see if it makes any difference.

dhalbert, 2024-06-18:
I cannot replicate the problem locally, so I want to try to figure out what's different about your setup.

Do you have any extensions installed? Could you try Chrome with no extensions? See https://www.howtogeek.com/779557/how-to-start-google-chrome-without-extensions-on-windows/

What are your DNS servers?
What version of Windows, and what version of Chrome?
Could you try Firefox, and Opera?

RetiredWizard, 2024-06-18:
@gitcnd Make sure you are not trying to reach https://cpy..... 

I thought I was seeing the same issue but google chrome was hiding the https://, by clicking on the URL line and removing the "s" so it was just http://cpy... things started working fine.

gitcnd, 2024-06-21:
@RetiredWizard @dhalbert 

Yes! (kindof) Looks like it might be an extension issue?  Chrome without extensions "half works" (and - weirdly - when working also fixes the other problem whereby any connection attempt from chrome locked up the serial console on the board making it unusable).  Chrome does *not* work in that same --disable-extensions browser in incognito mode however (the IP address did load, but not the mDNS one).  

The ESP32 required rebooting if connections failed (even the IP alone would not load it).

Another issue I noticed too:  after a fresh reboot and using it over serial manually from a terminal, then waiting a few hours, nothing could see the mDNS name (the IP did still work).  Rebooting the ESP32 fixed that, and the mDNS name could again be seen.

Manually disabling all my extensions got me a tiny bit further - but still not usable.  only a few of my typed character turn up:

[image]

@dhalbert what board and cpy version are you using?

Is there any easy way to change the MDN name?  it might be worth trying one with no hyphen (you never know...)

dhalbert, 2024-06-25:
> Is there any easy way to change the MDN name? it might be worth trying one with no hyphen (you never know...)

The hyphen shouldn't make any difference. There are plenty of host names with hyphens in them. Did you try another browser?

Does the serial console work well with a numeric IP? I wasn't sure whether that was working fine for you.

What DNS server are you using?
What kind of router arrangement are you using? Is the the ISP's router? Does it have unusual DNS settings?

 I have FiOS at home and I can also reach the board via something.fios-router.home. Your ISP may provide something similar.

gitcnd, 2024-06-26:
Numeric IP is fine.
DNS is 1.1.1.1 / 8.8.8.8  (which is irrelevant; mDNS doesn't use DNS, and http://cpy-eaf90c.local/ loads and works fine.)

Since http://cpy-eaf90c.local/ does work - I fail to see how anything else in my infrastructure could possibly be to "blame" - and even if it was - that's still a bug in the circuitpython/websocket code for not catering to it, since http://cpy-eaf90c.local/  **does** work?

dhalbert, 2024-06-26:
Maybe we are talking about two different meanings of "works fine". I meant to ask whether `/cp/serial` serial console works well with a numeric IP. Does `cp/serial` work well with any of the addresses you are using?

gitcnd, 2024-06-27:
Yes - /cp/serial works fine using an IP address; bearing in mind it is not serial - it's a websocket - this *really* needs to be renamed, because almost all boards have an _actual_ serial console, and calling 2 different things the same name violates the circuitpython "easy for beginners" paradigm

/cp/serial does not work, or is buggy/intermittent when using the mDNS name instead of the IP - same board, same network.

`http://192.168.1.118/`                     works
`http://192.168.1.118/cp/serial`       works
`http://cpy-eaf90c.local/`                 works
`http://cpy-eaf90c.local/cp/serial`    does not work properly, & is the wrong name (should be called and named /cp/websocket )

gitcnd, 2024-06-27:
update - I'll have to test some more boards - I'm getting some intermittent issues with http://192.168.1.118/cp/serial today as well.
I've tried an s2 mini (not my port) a lolin d1, and an esp32cam (my port) so far - I'll try some others...

aguaviva, 2024-06-28:
> update - I'll have to test some more boards - I'm getting some intermittent issues with http://192.168.1.118/cp/serial today as well. I've tried an s2 mini (not my port) a lolin d1, and an esp32cam (my port) so far - I'll try some others...

I am using your esp32cam (awesome work BTW!) and `/cp/serial` works only sometimes, even using the IP:

[image]

BTW you might want to check what [michalpokusa](https://github.com/michalpokusa) did, he added streaming to HTTPServer and it now can stream images: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/95  (I think this is stressing the camera code and I am getting some glitches once in a while)

dhalbert, 2025-07-29:
I'm going to close this since it's been over a year.

closed 2025-07-29 by dhalbert