adafruit/circuitpython #9339: Waveshare ESP32-S3-SIM7670G Support for CircuitPython

wraetz-dev, 2024-06-15:
Waveshare has a ESP32-S3R2 based board that has a SIM7670G-4G Cellular modem. It seems to share many elements with Waveshare's ESP32-S3-Pico board. They both have 2MB PSRAM and 16MB of flash memory. Wiki page for the ESP32-S3-SIM7670G is [here](https://www.waveshare.com/wiki/ESP32-S3-SIM7670G-4G).
The pin mapping appears different from ESP32-S3-Pico but I can't identify the relevant pins from the [pinout diagram](https://www.waveshare.com/img/devkit/ESP32-S3-SIM7670G-4G/ESP32-S3-SIM7670G-4G-details-inter.jpg) and [schematic](https://files.waveshare.com/wiki/ESP32-S3-A7670E-4G/ESP32-S3-A-SIM7670X-4G-Sch.pdf) provided by Waveshare.

This is a similar request to [8063](https://github.com/adafruit/circuitpython/issues/8063), and I have read the CircuitPython guide on adding a board--I just can't sort out which pins are which.

I am not affiliated with Waveshare in any way, just a hobbyist trying to make the board work with CircuitPython.

Can anyone point me in the right direction on adding the board?

tannewt, 2024-06-17:
The board addition guide is the right place to start. [Discord](https://adafru.it/discord) is a good place to ask for help implementing it.

kylefmohr, 2025-02-10:
Hi @wraetz-dev, have you made any additional progress at getting this board supported? 

I'd like to help, and I'll plan to start from scratch, but if you already have some of the work done, let me know.