adafruit/circuitpython #9341: ESP32S3: displayio causes "CircuitPython core code crash"

Sola85, 2024-06-16:
### CircuitPython version

```python
Main branch (see commit ids below) on espressif_esp32s3_devkitm_1_n8
```

### Code/REPL

```python
import busio, board
import displayio
displayio.release_displays()
# The issue is reproducable whether there is something connected to these pins or not. The exact pins used here also don't matter.
spi = busio.SPI(board.IO5, board.IO4, None)
display_bus = displayio.FourWire(spi, command=board.IO3, chip_select=board.IO2, reset=board.IO1)

print("\nDone setup")
```

### Behavior

```
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
```
This occurs after saving the above code in `code.py`. Sometimes the crash only occurs after saving the file a few (usually < 10) times.

### Description

At https://github.com/adafruit/circuitpython/commit/55028d0a2d52a1de191657e7893ec11d1bb7b952, everything was working normally, but at https://github.com/adafruit/circuitpython/commit/ed5591c5cbc6fc26125c5f403de4bff2f9eac4ee (94 commits later) I am seeing this issue.

Between these two versions, I am also seeing a significant increase in power consumption (previously 50-60mA while in repl, now 90-100mA)

If this is known to be related to some work that is still in progress (this is not a release version of CircuitPython after all), feel free to close the issue.

dhalbert, 2024-06-25:
Hi - The code above runs and exits, leaving you in the REPL. Is there more, like a `while True: pass` loop? Saving `code.py`  will not restart code.py if you are already in the REPL.

I did add a `while True:` to the above and tried it on a QT Py S3 8/0 board, changing some pin names to make it work, and also on an ESP32-S3 DevKitC 8/8 board. (I don't have the similar DevKitM board.) I couldn't make a safe mode crash with the tip of main at the moment. I repeatedly edited code.py slightly to make it restart.

Sola85, 2024-06-26:
No, I did not have a `while True:` afterwards. After the code finished running I got the message 
```
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
(assuming the crash was not triggered) and the code would automatically re-run if I saved the code.py again. Saving a few times would eventually (but reliably) trigger the crash.

However, I just retested in the current main branch (Commit ac598a399a2e557f905c16c8793e15f6d4db1139) and can't reproduce it anymore either (neither with the above minimal example, nor my actual use case). 
Hence I'll close this issue and re-open in case I see it again.

closed 2024-06-26 by Sola85