adafruit/circuitpython #9355: collections.deque status not matching documentation

mMerlin, 2024-06-20:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.5 on 2024-05-22; Adafruit PyPortal with samd51j20
```

### Code/REPL

```python
from collections import deque
a = deque((), 5)
len(a)
a.append(0)
a.append(1)
a.append(2)
a.append(3)
a.append(4)
len(a)
a[0]
list(a)
```

### Behavior

```text
0
5
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'deque' object isn't subscriptable
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'deque' object is not iterable
```

### Description

The circuitpython collection.deque package does not match the functionality described in https://docs.circuitpython.org/en/latest/docs/library/collections.html#collections.deque
Either the documentation is wrong, or deque is broken, at least for the pyportal build.

### Additional information

The information in older issues seem to indicate that deque was implemented, including making it iterable, but that is not what I see. I see a few deque references in the Adafruit Learning System Guides repo:
grep -R --include="*.py" 'deque' .
but from my tests only one of them would actually work in this environment. The empty list with limit. Tests show that using a list for the iterable parameter does not work, and neither does specifying the maxlen keyword.
Another search in the circuitpython repo found some references in tests. Again, many would appear to fail from the repl tests I did. The comment in tests/cpydiff/modules_deque.py says "Deque not implemented", which is not accurate either. A hold over from the fork?

dhalbert, 2024-06-20:
We inherit the implementation from MicroPython. There are separate `MICROPY_PY_COLLECTIONS_DEQUE_SUBSCR` and `MICROPY_PY_COLLECTIONS_DEQUE_ITER` flags, which should be on, but apparently are not. I'll investigate.

dhalbert, 2024-06-20:
Aha, this was fixed already by #9065. The new functionality is in 9.1.0-beta.3 and later. Looking at the `latest` documentation a bit confusingly tells you about 9.1.0-beta.3, not 9.0.5

> tests/cpydiff/modules_deque.py says "Deque not implemented"

This is a problem from upstream MicroPython, I think.

closed 2024-06-20 by dhalbert

dhalbert, 2024-06-20:
> > tests/cpydiff/modules_deque.py says "Deque not implemented"
> 
> This is a problem from upstream MicroPython, I think.

Looking more closely at this, I think that test is invoked only when `deque` is turned off.