adafruit/circuitpython #9360: USB audio source/sink

elipsion, 2024-06-21:
I've been tinkering around with the audio infrastructure and, while it's fun to do stuff like playing WAV files of an SD card, am kind of missing the functionality to get audio streams to and from my computer over USB.

Ideally I'd like to be able to expose a number of sources/sinks to the host and perform mixing on them together with the `audiobusio` module.

tannewt, 2024-06-21:
That'd be neat! Our audio subsystem is pretty complicated but would benefit from USB support. This be especially useful on USB high speed devices.

tannewt, 2024-06-21:
TinyUSB already has audio support IIRC so this would involve hooking that into CP's audio subsystem.

elipsion, 2024-06-21:
> TinyUSB already has audio support IIRC so this would involve hooking that into CP's audio subsystem.

I did check that before I submitted the request, to not put in an unreasonable FR.

Is there anything I can do to help with implementation? I realize it might be a bit of a tough nut to crack as a first contribution, but I would gladly help out with testing or documentation.

tannewt, 2024-06-24:
I don't know of anyone who will be implementing it now. So, it's up to you. We can guide you here or [on Discord #circuitpython-dev](https://adafru.it/discord) but it'll be up to you to implement.